ARMmbed/mbed-os #10237: Compile secure code for PSA test

ccli8, 2019-03-27:
### Description

Based on #9221, I am adding PSA support to Nuvoton's target **NUMAKER_PFM_M2351** which is Coretex-M23 based. To verify my porting, I try to run PSA tests and get PSA test result below. For failed items like `mbed-os-tests-psa-crypto_access_control`, I find my secure code misses necessary PSA test partition and secure-side code in **mbed-os/TESTS/psa/crypto_access_control**. Then how to compile my secure code to support PSA tests?

```
| NUMAKER_PFM_M2351_NS-ARMC6 | NUMAKER_PFM_M2351 | mbed-os-tests-psa-attestation                                                        | OK          | 25.48              | default     |
| NUMAKER_PFM_M2351_NS-ARMC6 | NUMAKER_PFM_M2351 | mbed-os-tests-psa-crypto_access_control                                              | TIMEOUT     | 138.08             | default     |
| NUMAKER_PFM_M2351_NS-ARMC6 | NUMAKER_PFM_M2351 | mbed-os-tests-psa-crypto_init                                                        | OK          | 17.97              | default     |
| NUMAKER_PFM_M2351_NS-ARMC6 | NUMAKER_PFM_M2351 | mbed-os-tests-psa-entropy_inject                                                     | OK          | 19.1               | default     |
| NUMAKER_PFM_M2351_NS-ARMC6 | NUMAKER_PFM_M2351 | mbed-os-tests-psa-its_ps                                                             | OK          | 18.25              | default     |
| NUMAKER_PFM_M2351_NS-ARMC6 | NUMAKER_PFM_M2351 | mbed-os-tests-psa-spm_client                                                         | TIMEOUT     | 80.78              | default     |
| NUMAKER_PFM_M2351_NS-ARMC6 | NUMAKER_PFM_M2351 | mbed-os-tests-psa-spm_server                                                         | SYNC_FAILED | 44.39              | default     |
| NUMAKER_PFM_M2351_NS-ARMC6 | NUMAKER_PFM_M2351 | mbed-os-tests-psa-spm_smoke                                                          | TIMEOUT     | 38.8               | default     |
```

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

@mikisch81

mikisch81, 2019-03-27:
All the failed tests have dedicated secure binaries which are different from the default binary by that they have more partitions which handle the test logic.

Is this PSA work going to be in a PR so we can see the entire picture?

cmonr, 2019-03-27:
@orenc17, though I think I know the answer (but it's a bit late here)

ccli8, 2019-03-27:
This PSA work will run after #9221 and will become a PR finally. Would you tell me how to compile secure code to run PSA test?

mikisch81, 2019-03-27:
It would be more helpful if we move the discussion to some kind of a private fork of your work, maybe with an internal PR there which will show the changes relative to master.
You are actually adding a new target to Mbed-OS (m2351 PSA target) so I think the "Issues" section here is not the best interface for it.
Another thing, there is already a v8m PSA target in Mbed-OS - [NXP LPC55S69](https://github.com/ARMmbed/mbed-os/tree/master/targets/TARGET_NXP/TARGET_MCUXpresso_MCUS/TARGET_LPC55S69), so you can look at there for reference as well, and there is the [tf-m-integration](https://github.com/ARMmbed/mbed-os/blob/master/components/TARGET_PSA/TARGET_TFM/tf-m-integration.md).

And for your specific question, to build the secure images (after you defined a new secure target in target.json), see [here](https://github.com/ARMmbed/mbed-os/tree/master/targets/TARGET_NXP/TARGET_MCUXpresso_MCUS/TARGET_LPC55S69/TARGET_M33_NS/prebuilt) for reference:

## tfm.bin

```sh
mbed compile -t ARMC6 -m NUMAKER_PFM_M2351_S --app-config mbed-os/tools/psa/tfm/mbed_app.json --profile release
```

## spm_smoke.bin

```sh
mbed test --compile -t ARMC6 -m NUMAKER_PFM_M2351_S --app-config mbed-os/tools/psa/tfm/mbed_app.json -n mbed-os-tests-psa-spm_smoke -DUSE_PSA_TEST_PARTITIONS -DUSE_SMOKE_TESTS_PART1 --profile release
```

## spm_client.bin

```sh
mbed test --compile -t ARMC6 -m NUMAKER_PFM_M2351_S --app-config mbed-os/tools/psa/tfm/mbed_app.json -n mbed-os-tests-psa-spm_client -DUSE_PSA_TEST_PARTITIONS -DUSE_CLIENT_TESTS_PART1 --profile release
```

## spm_server.bin

```sh
mbed test --compile -t ARMC6 -m NUMAKER_PFM_M2351_S --app-config mbed-os/tools/psa/tfm/mbed_app.json -n mbed-os-tests-psa-spm_server -DUSE_PSA_TEST_PARTITIONS -DUSE_SERVER_TESTS_PART1 -DUSE_SERVER_TESTS_PART2 --profile release
```

## crypto_access_control.bin

```sh
mbed test --compile -t ARMC6 -m NUMAKER_PFM_M2351_S --app-config mbed-os/tools/psa/tfm/mbed_app.json -n mbed-os-tests-psa-crypto_access_control -DUSE_PSA_TEST_PARTITIONS -DUSE_CRYPTO_ACL_TEST --profile release
```

`tfm.bin` is the default image which most of the test binaries should use (or any external application), the other binaries are specific for those tests, as those tests need additional secure partitions.

ccli8, 2019-03-27:
@mikisch81 Many thanks for your information on how to compile secure code for PSA test. Internal PR is a good suggestion for discussion. I will raise one when necessary.

closed 2019-03-27 by ccli8

mikisch81, 2019-03-27:
@ccli8 please tag @ARMmbed/mbed-os-psa team in the upcoming PR