ARMmbed/mbed-os #10238: TMPM46B does not support `DigitalOut(NC)` instanciation.

ithinuel, 2019-03-27:
### Description

The AdBun-M46B (TMPM46B) target does not support instantiation of `DigitalOut` ios when the corresponding pin is `NC` (typically an unused SPI CS or Reset pin in a driver).

This is triggered by the MBED_ASSERT from `gpio_set` called [gpio_api.c#L40](https://github.com/ARMmbed/mbed-os/blob/7e7f4f561bcdf84e50c947ebd2d5a14437f36f1c/targets/TARGET_TOSHIBA/TARGET_TMPM46B/gpio_api.c#L40).

`gpio_init` shall return transparently and the rest of the API shall ignored any NC pin.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-04-03:
cc @ARMmbed/team-toshiba

ganesh-ramachandran, 2019-04-04:
@ithinuel, created PR #10317 for this issue.

cmonr, 2019-04-08:
Closing since PR was merged. Feel free to reopen if the PR didn't solve this issue.

closed 2019-04-08 by cmonr