ARMmbed/mbed-os #10239: Ethernet rx packet size issue on NXP i.MX RT1050

cyborg71, 2019-03-27:
### Description

We have an issue of receiving IP packets larger than 1490 bytes over Ethernet on the NXP i.MX RT1050 target. The packet size 1490 bytes includes Ethernet payload data excluding Ethernet frame header. After a few or occasionally dozens of sucessfully received IP packets with size above 1490 bytes we have seen that the IP layer stops receiving packets. When this issue occurs it is not possible to receive any more IP packets of any size over Ethernet.

We are sending data in a very slow speed (1 packet/second) to the i.MX RT1050 target when seing this issue. We have never seen this issue when receiving IP packets with size <= 1490 bytes.  

When debugging the Ethernet drivers (fsl_enet.c and imx_emac.cpp) when the RX issue occurs we have seen some different issues in the drivers. Most of the times we have seen that the allocation of new buffers continously fails in the low_level_input() function in imx_emac.cpp. The system will never recover from this. Each received Ethernet frame continously generates a failure to allocate a new buffer with size 1522 bytes.

imx_emac.cpp:
```
emac_mem_buf_t *Kinetis_EMAC::low_level_input(int idx)

        /* Attempt to queue new buffer */
        temp_rxbuf = memory_manager->alloc_heap(ENET_ETH_MAX_FLEN, ENET_BUFF_ALIGNMENT);
        if (NULL == temp_rxbuf) {
            /* Re-queue the same buffer */
            update_read_buffer(NULL);

#ifdef LOCK_RX_THREAD
      TXLockMutex.unlock();
#endif
      	    emacInputReturnNull++;
            return NULL;
        }
```

Another observation when the Ethernet RX issue occurs is that instead of the above buffer allocation problem the RX interrupt routine **ENET_ReceiveIRQHandler()** in **fsl_enet.c** stops to execute. As a consequence the ENET_IEEE_R_MACERR register is increased for each received ETH frame and the frame is never read out by the Ethernet drivers.  We have also seen some random occurences of hard faults when receiving IP packets with size > 1490 bytes.

We can reproduce this issue after every reboot by pinging an "i.MX RT1050" target with data size > 1462 bytes (+ ICMP header 8 bytes and IP header 20 bytes).
`# ping -s 1463 192.168.1.10`

Also easily reproducable by sending messages over UDP to an "i.MX RT1050" target with data size > 1462 bytes (+ UDP header 8 bytes and IP header 20 bytes) and having a simple application receiving data on this UDP port. We are sending data in a very slow speed to reproduce the problem. The "i.MXRT1050" fails to handle incoming packets after a few or occasionally dozens received packets over Ethernet. We dont see the RX issue when sending IP packets with a size <= 1490 bytes.

Tested with mbed-os 5.11.1 and 5.12.0 on IMXRT1050-EVKB evaluation board from NXP with hyperflash and our production "i.MX RT1050" target with QSPI NOR flash.

Compiled with gcc-arm-none-eabi-6-2017-q2-update and gcc-arm-none-eabi-7-2018-q2update

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

unsignedint, 2019-03-27:
@mmahadevan108 it would be excellent if you could please take a look at this :)

mmahadevan108, 2019-03-28:
Can you send me the code for the test application running on MXRT when pinging.

cyborg71, 2019-03-28:
[main.zip](https://github.com/ARMmbed/mbed-os/files/3018651/main.zip)

mmahadevan108, 2019-03-28:
Thank you, did you accidentally close the issue?

cyborg71, 2019-03-28:
Sorry. My mistake. Reopened the issue again.

mmahadevan108, 2019-03-28:
@SeppoTakalo @kjbracey-arm . Preliminary debugging seems to indicate the crash is occurring in the LWIPMemoryManager. Can you help take a look at this on the RT1050.

mmahadevan108, 2019-03-28:
Below is the PC reported by the mbed fault handler
++ MbedOS Fault Handler ++
FaultType: HardFault

PC   : 60004B28

I did a lookup of this address:
$ arm-none-eabi-readelf.exe -a BUILD/MIMXRT1050_EVK/GCC_ARM/mymbed.elf | grep 60004b
   317: 60004b0a     0 NOTYPE  LOCAL  DEFAULT    4 $t
   318: 60004b30     0 NOTYPE  LOCAL  DEFAULT    4 $t
   319: 60004b36     0 NOTYPE  LOCAL  DEFAULT    4 $t
   320: 60004b6a     0 NOTYPE  LOCAL  DEFAULT    4 $t
   321: 60004b8e     0 NOTYPE  LOCAL  DEFAULT    4 $t
   324: 60004bb8     0 NOTYPE  LOCAL  DEFAULT    4 $t
   325: 60004bc4     0 NOTYPE  LOCAL  DEFAULT    4 $t
   326: 60004bc6     0 NOTYPE  LOCAL  DEFAULT    4 $t
   327: 60004bd6     0 NOTYPE  LOCAL  DEFAULT    4 $t
   328: 60004bd7    52 FUNC    LOCAL  DEFAULT    4 _ZL25convert_mbed_addr_to
  3177: 60004b37    52 FUNC    GLOBAL DEFAULT    4 _ZN17LWIPMemoryManager12a
  3212: 60004b0b    38 FUNC    GLOBAL DEFAULT    4 _ZN17LWIPMemoryManager13s
  3226: 60004b6b    36 FUNC    GLOBAL DEFAULT    4 _ZN17LWIPMemoryManager10a
  3409: 60004b8f    42 FUNC    GLOBAL DEFAULT    4 _ZN17LWIPMemoryManager10a
  3833: 60004bc5     2 FUNC    WEAK   DEFAULT    4 _ZN4mbed8CallbackIFiiNS0_
  3989: 60004bc7    16 FUNC    GLOBAL DEFAULT    4 _ZN4LWIP14tcpip_init_irqE
  4248: 60004b31     6 FUNC    GLOBAL DEFAULT    4 _ZN17LWIPMemoryManager7se
  4272: 60004bb9    12 FUNC    WEAK   DEFAULT    4 _ZN4mbed8CallbackIFiiNS0_

mmahadevan108, 2019-04-02:
@maclobdell @kjbracey-arm @SeppoTakalo Do you have any suggestions?

kjbracey-arm, 2019-04-02:
Presumably the failing function is `LWIPMemoryManager::set_total_len`, so I'm guessing that the `pbuf` has been corrupted, or the pointer to it.

But I can't see how that would occur. The descriptors, packet buffers, and the rx_buff array should all be miles away from each other, so any likely buffer overrun from hardware doesn't seem likely to cause a problem that easily.

I've stared at all the buffer size and alignment maths, and I can't see any problems so far. The driver is almost identical to the K64F one, which we know is solid.

I know the K64F doesn't have to contend with Cortex-M7 caching though, so maybe there's a problem in that area, but the descriptors are supposed to be uncached anyway.

Can you get a backtrace of the fault? Look into some of the rx buffer pointers?

mmahadevan108, 2019-04-02:
Thank you. I am unable to reproduce the failure on K66F which is similar to K64F.

kjbracey-arm, 2019-04-02:
Ah, I guess the actual lwIP pbuf structure is in the heap adjacent to the actual data payload. If there was some sort of cache cleaning problem in the underlying driver, it might be destroying a pbuf header.

Looking at fsl_enet.c, I see some `DCACHE_InvalidateByRange` calls. That could be destroying data near the buffer, as it only has cache line (32-byte) granularity.

But then, are the ENET calls containing those actually used by the driver? Doesn't seem like they are.

But then, if they're not, who IS doing cache maintenance on the buffers? Anyone?

What I'd expect to see to have this working solidly is the RX buffers being 32-byte aligned to avoid interfering with neighbouring data (or being interfered by). You need a "dcache invalidate" somewhere in the reception path, and you need to make sure that the invalidate doesn't wipe someone else's data, and that if someone modifies neighbouring data, it doesn't end up writing back a cache line over your packet.

Transmission doesn't pose any alignment issues - you just need to clean, without invalidating.

kjbracey-arm, 2019-04-02:
Or of course it doesn't need to be anything to do with the cache, it could be straight data overrun in the heap. Have you tried globally disabling the data cache?

mmahadevan108, 2019-04-02:
Thank you @kjbracey-arm . I have made changes to the MXRT1050 ENET driver, PR is above. 

@cyborg71 Can you give this a try. I did not see any issues when running the ping test.

cyborg71, 2019-04-03:
@mmahadevan108: Thanks! I have tried the fix from the above PR and it works to receive large packets. Tried both ping and UDP packets. Could not see any other issues in our system when doing some initial tests.

kjbracey-arm, 2019-04-03:
@cyborg71 Can you identify which part(s) of the change actually make the difference? I have a suspicion it's just the oversized buffer, rather than the cache thing.

Are you sure the data cache is actually enabled?

Actually, looking at mbed_overrides.c, it seems like the main memory region (80000000-81E00000) is marked shareable, which does deactivate any caching on the Cortex-M7.

So at the minute, the cache operations should have no effect. You would need to do a bunch of cache work if you wanted to mark those non-shareable.

mmahadevan108, 2019-04-03:
@cyborg71 @kjbracey-arm . The tests passed without the cache operations and just the ENET buffer size fix. I will update the PR for your testing.

kjbracey-arm, 2019-04-03:
The thing is, it's not clear why the "buffer size fix" is a fix.  What's the actual problem here? The frame length limit is set to 1522, and both heap_alloc and the ENET_ALIGN macro should be independently rounding that up to 1524, so that we have 1524-byte buffers allocated, and are telling the fsl_enet.c that we have 1524 byte buffers.

Your addition ends up adding a bit more slack, because of the way the maths works in heap_alloc, but I don't see why the slack is necessary. It suggests someone is getting maths wrong somewhere else.

mmahadevan108, 2019-04-03:
ENET_ALIGN is aligning to a 64-byte boundary.

mmahadevan108, 2019-04-03:
One observation, I see the below line adds the alignment provided to the size. 
https://github.com/ARMmbed/mbed-os/blob/master/features/lwipstack/LWIPMemoryManager.cpp#L21
However the size is not aligned, with the change to pass aligned size, the resulting buffer allocated is also aligned.

kjbracey-arm, 2019-04-03:
Good, I was just realising from the manual that it was supposed to be 64-byte. Looks like my IDE was finding the wrong definition of `ENET_BUFF_ALIGNMENT`.

And I've just realised I was misunderstanding `alloc_heap`. It only ensures alignment of the buffer, not the actual data block. You do indeed need to round up your size, as you do need a 1536-byte block that is 64-byte aligned. You're actually being given a 1522-byte block that is 64-byte aligned, which is no good.

It's possible a number of drivers may need this change. I don't think `alloc_heap` is wrong - in some cases there may be an alignment requirement for the start address which is stricter than the actual length.

mmahadevan108, 2019-04-03:
@cyborg71 . Could you kindly try the update PR

cyborg71, 2019-04-03:
@mmahadevan108 I will try the updated PR right now.

cyborg71, 2019-04-03:
@mmahadevan108 I have now tested the updated PR and it works to recieve large packets, both ping and UDP.

closed 2019-04-09 by ciarmcom