ARMmbed/mbed-os #10241: Dynamic memory allocation in mbed-os

desowin, 2019-03-27:
### Description

I have noticed that [mbed-os/components/storage/blockdevice/COMPONENT_QSPIF/QSPIFBlockDevice.cpp](https://github.com/ARMmbed/mbed-os/blob/899ea5991cfe23040bd7c766fde7fa667f4b8c41/components/storage/blockdevice/COMPONENT_QSPIF/QSPIFBlockDevice.cpp#L108) does dynamically allocate the QSPIFBlockDevice::_active_qspif_flash_csel_arr array. Would it be possible to allocate this array statically?

If a target has very little RAM left and there is basically no room for heap, it is benefitical if the build failed at linking time as opposed to at runtime. Thus I would prefer if mbed-os components used only statically allocated memory.

Is there any guideline/rule regarding dynamic memory use in mbed-os core?

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

cmonr, 2019-03-27:
@ARMmbed/mbed-os-storage

offirko, 2019-03-28:
@desowin - Thanks for your query.
Because this 40 bytes global array required initializing different than '0's it was allocated and initialized dynamically. Also, because it is a global variable, allocating and initializing will be done before 'main' . Still, the point you raise is valid and if critical I can move it to be statically allocated.

dannybenor, 2019-03-28:
Always is better to allocate statically if possible

offirko, 2019-03-28:
Internal bug created: https://jira.arm.com/browse/IOTSTOR-819

offirko, 2019-04-01:
Fixed with: https://github.com/ARMmbed/mbed-os/pull/10283

desowin, 2019-04-18:
Could the reason for closing this be public? The QSPIFBlockDevice.cpp was not modified to address this issue.

offirko, 2019-04-18:
In the long term QSPIFBlockDevice should support several instances on the same csel, so the "_active_qspif_flash_csel_arr array" would no longer be required then. Current impact on dynamic heap allocation is minimal (only 40 bytes).

@cmonr  - please advise regarding change priority and owner

closed 2019-10-31 by ciarmcom