ARMmbed/mbed-os #10242: Mbed error info not printed with mbed-os-5.12.0

desowin, 2019-03-27:
### Description

Consider following dummy application:
```
int main(void)
{
    char *buf;
    do
    {
        buf = new char[1024];
    }
    while (buf);
}
```

As the application leaks memory, mbed error will eventually occur and on serial port there'll be information like (mbed-os-5.11.5):
```
++ MbedOS Error Info ++
Error Status: 0x8001011F Code: 287 Module: 1
Error Message: Operator new[] out of memory

Location: 0x80029A5
Error Value: 0x400
Current Thread: main  Id: 0x2000551C Entry: 0x8002EA3 StackSize: 0x1000 StackMem: 0x20005560 SP: 0x200064F4
For more info, visit: https://armmbed.github.io/mbedos-error/?error=0x8001011F
-- MbedOS Error Info --
```

However with mbed-os-5.12.0, no data is received on serial port.

I am observing the issue with 32F746GDISCOVERY board. The command I use to compiile and run the application is `mbed compile --target DISCO_F746NG -t GCC_ARM --flash`

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

cmonr, 2019-03-27:
...huh. That's an interesting test case.

CC @ARMmbed/mbed-os-core

kegilbert, 2019-03-27:
Tested on a K64F with GCC_ARM on 5.11, 5.12, and master and was able to get 
```
++ MbedOS Error Info ++
Error Status: 0x8001011F Code: 287 Module: 1
Error Message: Operator new[] out of memory
...
```

each time with the described code snippet. I'll try to track down a 32F746GDISCOVERY, might be something target specific.

cmonr, 2019-04-09:
> However with mbed-os-5.12.0, no data is received on serial port.

I recall seeing another issue about serial port intiialization. I wonder if this is related.

cmonr, 2019-04-09:
Possibly related: https://github.com/ARMmbed/mbed-os/issues/10344

PR #10358 (2019-04-17): Error print improvements
### Description

Fixes two key problems:
* `mbed_error` or `MBED_ASSERT` failure could fail to output if the console had not previously been used
* `mbed_error` or `MBED_ASSERT` failure would be garbled if `platform.stdio-buffered-serial` was `true` but `platform.stdio-convert-newlines` was `false`.

Getting the console initialized for the first time in a fatal crash situation is always going to be potentially dicey, but we can make a better effort.

(If a platform is really serious about getting crash information out of the serial port, they should make an effort to g [... cut]
closed 2019-04-17 by 0xc0170