ARMmbed/mbed-os #10245: Reading SD Card returns wrong 0x00h data value - CY8CKIT_062_WIFI_BT

offirko, 2019-03-27:
### Description
Target: CY8CKIT_062_WIFI_BT

Reading SD Card will sometime return 0x00h byte value instead of real value, while another thread runs in the background (the second thread has very low CPU usage).

Note that: SD Read brings CPU Usage to 100%

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

offirko, 2019-03-27:
PSB attached log.
Test scenario:
1) Program 4MB of data to SD Card .
2) Program Done
3) Start CPU Measurement and working thread (~3% CPU)
4) Start Reading 4MB Data at 1KB chunks <== CPU rises to 100%
5) Several reads return byte value 0x00h instead of actual data
6) Read Done <== CPU drops back to 3%

[CY8CKIT_062_WIFI_BT_SD_Read_with_work_thread.txt](https://github.com/ARMmbed/mbed-os/files/3013869/CY8CKIT_062_WIFI_BT_SD_Read_with_work_thread.txt)

offirko, 2019-03-27:
CC: @dannybenor , @maclobdell

alzix, 2019-03-27:
@offirko, can you upload your test project to GitHub and share the link here?
It will help reproducing the issue.

offirko, 2019-03-27:
@alzix : sure thing..
test name: "features-storage-tests-blockdevice-general_block_device"
test suite:  "SD Testing prog read mult"

Repo:
******
https://github.com/offirko/mbed-os/tree/offir_CY8CKIT_062_WIFI_BT_sd_read

Build:
******
mbed test -v -t GCC_ARM -m CY8CKIT_062_WIFI_BT -n features-storage-tests-blockdevice-general_block_device --compile --app-config mbed_app.json -DMBED_CPU_STATS_ENABLED

Run Test:
**********
mbed test -v -t GCC_ARM -m CY8CKIT_062_WIFI_BT -n features-storage-tests-blockdevice-general_block_device --run -v

maclobdell, 2019-05-17:
@offirko  - I tried out your test on CY8CKIT_062_WIFI_BT and I do see some failures reported in the form 
```
mbedgt: :390::FAIL: Expected 222 Was 0
```
However, I also tried it on FRDM-K64F as a reference and I see massive failures.  I'm using the same sd card shield.   I just swapped the MCU board.  I have not dug into it more.  Any suggestion?

offirko, 2019-05-19:
@maclobdell - line 390 where CY8CKIT_062_WIFI_BT failed , is comparing the read data to the programmed data. We saw with CY8CKIT_062_WIFI_BT  that CPU reached to 100% on SD card IO.
I would suggest checking whether CPU reaches 100% on K64F, and see what are the specific failures for K64F.

maclobdell, 2019-06-19:
@offirko  I started looking at this again.  I ran the tests on the Cypress board.  I noticed that the failing position is always at the 1022, 1023, 510, 511.  

The block size is set to 4096 when writing.
The block size is set to 1024 when reading.  
The 512 byte program size.  

Is this of any significance to you? 

Pulling out the failure string and sorting it, I get this.  
The failure is always reading 00's in the last two bytes of some blocks.  

pos: 1022, exp: 6f, act:   00
pos: 1022, exp: a5, act: 00
pos: 1022, exp: b0, act: 00
pos: 1023, exp: 84, act: 00
pos: 1023, exp: a1, act: 00
pos: 1023, exp: c4, act: 00
pos: 1023, exp: dd, act: 00
pos: 1023, exp: f3, act: 00
pos: 1023, exp: f6, act: 00
pos: 1023, exp: f9, act: 00
pos: 1023, exp: fd, act: 00
pos: 510, exp: 64, act: 00
pos: 510, exp: da, act: 00
pos: 511, exp: 10, act: 00
pos: 511, exp: 31, act: 00
pos: 511, exp: 76, act: 00
pos: 511, exp: 7c, act: 00
pos: 511, exp: 93, act: 00
pos: 511, exp: c0, act: 00
pos: 511, exp: d7, act: 00
pos: 511, exp: eb, act: 00
pos: 511, exp: fe, act: 00

The failing blocks are: 

Assert Failed Buf Read -   block:size: 4076000:1024
Assert Failed Buf Read - block:size:   40ea000:1024
Assert Failed Buf Read - block:size:   40ea000:1024
Assert Failed Buf Read - block:size:   40fe800:1024
Assert Failed Buf Read - block:size:   40fe800:1024
Assert Failed Buf Read - block:size:   4108c00:1024
Assert Failed Buf Read - block:size:   410d400:1024
Assert Failed Buf Read - block:size:   415cc00:1024
Assert Failed Buf Read - block:size:   420f000:1024
Assert Failed Buf Read - block:size:   423a400:1024
Assert Failed Buf Read - block:size:   423a400:1024
Assert Failed Buf Read - block:size:   42d2800:1024
Assert Failed Buf Read - block:size:   42fc800:1024
Assert Failed Buf Read - block:size:   42fd000:1024
Assert Failed Buf Read - block:size:   4327400:1024
Assert Failed Buf Read - block:size:   4327400:1024
Assert Failed Buf Read - block:size:   435ac00:1024
Assert Failed Buf Read - block:size:   4370000:1024
Assert Failed Buf Read - block:size:   43b0400:1024
Assert Failed Buf Read - block:size:   43b0400:1024
Assert Failed Buf Read - block:size:   43ccc00:1024

Please help!

maclobdell, 2019-06-19:
I'm curious if this is of significance.  the clock keeps going for two sequences after the chip select is deasserted.  
[image]

40Grit, 2019-06-19:
Mac have you tried this with the proto? I was going to test SD on it sometime this week or next.

offirko, 2019-06-20:
@maclobdell - my initial analysis showed that  SD read issues started only when CPU reaches 100%. Once that point is reached its rather unpredictable what will go wrong. I believe hal analysis and perhaps also HW analysis are required. Since I have switched positions a few months ago, I'll CC the current storage and client people that may assist with further analysis:
1. Storage : @SeppoTakalo 
2. Client: @jenia81

maclobdell, 2019-06-20:
@40Grit  I have not tested SD on the proto board.  It will likely be affected by this issue.

maclobdell, 2019-06-20:
I ran the test case on https://github.com/ARMmbed/mbed-os/pull/10692.  It also fails.  So, even with the HAL updates from Cypress, this issue persists.

maclobdell, 2019-06-20:
I've found that the failures vary greatly depending on what the worker tasks are doing.  Specifically, I see more failures, the more the worker task is blocking.  If in the worker task, I add: 
```
while(1) {
Thread::wait(1);
}
```  
Then lots of failures.  That seems opposite of what I would expect.  A waiting task should block and allow other tasks to run, thus getting out of the way.

maclobdell, 2019-11-13:
Closing due to no activity and low priority.  Will re-open if SPI failures are encountered in the future.

closed 2019-11-13 by maclobdell