ARMmbed/mbed-os #10247: Online Compiler - SoftDevice not being included in Nordic builds

TacoGrandeTX, 2019-03-27:
### Description

A customer noticed that a nRF52-DK project based on v5.10 is no longer including the SoftDevice after the v5.12 tools update to the Online Compiler.

I confirmed that and have a legacy v5.9.1 project that can be used for testing:
* https://os.mbed.com/users/RalphF/code/OS5_BLE_LED/

It's possible that we're just outputing the application hex file as opposed to the combined hex as the generated file contains "NRF52_DK_application".  The addresses in the hex file sure are strange.  Attaching the hex file with a *.txt extension.

[OS5_BLE_LED.NRF52_DK_application.txt](https://github.com/ARMmbed/mbed-os/files/3015520/OS5_BLE_LED.NRF52_DK_application.txt)

 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

TacoGrandeTX, 2019-03-28:
We have received another report from the field that this also impacts nRF51 projects.

theotherjimmy, 2019-03-28:
I think the NRF51 stuff is fixed by https://github.com/ARMmbed/mbed-os/pull/10065

bsturgess, 2019-03-28:
Any further updates/fixes on this for NRF52832?    Starting to get desperate here!

cmonr, 2019-03-28:
@bsturgess Would you mind elaborating on what specifically you're getting desperate about, or what "this" is in reference to?

bsturgess, 2019-03-28:
@cmonr not being able to build any projects, stopping all production here.

cmonr, 2019-03-28:
@bsturgess Please open a new issue, since it sounds like this is an unrelated issue.

bsturgess, 2019-03-28:
No this is the same issue, opened by Ralph, of whom I spoke with on email to make him aware last night.

Really not sure where the confusion is coming from? The online compiler was broken for NRF targets when the 'tools upgrade' for 5.12 was deployed yesterday morning, since then, anyone with NRF52 or 51 targets cannot run binaries due to missing softdevice.

Also, here are 3 related posts from people experiencing the same issue.

https://os.mbed.com/questions/85287/The-bin-file-from-mbed-compiler-Stopped-/?c=33022
https://os.mbed.com/questions/85295/Online-compiler-for-NRF51822-no-longer-w/
https://os.mbed.com/questions/85311/Compilation-error/

cmonr, 2019-03-28:
@bsturgess Ah, I see.

The missing link was that neither you or @TacoGrandeTX mentioned that there had been some communication about this problem, outside of this issue. The context was missing from my end.

bsturgess, 2019-03-28:
@cmonr No worries, can you confirm the issue is being investigated as a priority?

cmonr, 2019-03-28:
@bsturgess I can confirm that the issue is being actively looked into.

cmonr, 2019-03-28:
Since this appears that it's already being looked into, I'm going to recluse myself from this particular issue.

bridadan, 2019-03-28:
We have a fix ready, just waiting on deployment.

TacoGrandeTX, 2019-03-29:
Fix was deployed this morning UK time.

bsturgess, 2019-03-29:
Thanks all, much appreciated. We are back in action

closed 2019-04-01 by ciarmcom