ARMmbed/mbed-os #10249: Tools: Mbed OS scripts are not fully python 3 compatible

OPpuolitaival, 2019-03-28:
### Description

Python 3 should be fully supported for a while already. Still there are a lot of python 3 incompatible code in mbed-os repository
```
2to3 -n -w tools/
git status | wc -l
```
Gives 127 files needing changes. All of those are not relevant but there are even incompatible prints.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

theotherjimmy, 2019-03-28:
@OPpuolitaival `git status | wc -l` does not equal the files changed. `git status --porcelain | wc -l` does, and gives me 118 files changed. not quite as bad.

theotherjimmy, 2019-03-28:
it also deletes:
```python
from __future__ import print_function, division, absolute_import
```
which is needed for python 2 compatibility.

theotherjimmy, 2019-03-28:
FYI, the `tools/compliance` and `tools/dev` directories are dead code to my knowledge. Confirmed with grep.

theotherjimmy, 2019-03-28:
`build_release.py` needed 2to3.

theotherjimmy, 2019-03-28:
@OPpuolitaival What I'm trying to get at is that your metric does not serve it's purpose: it does not gauge our python3 readyness.

cmonr, 2019-03-28:
Commenting here since this is public.

Taking a look at what `2to3` does, as has already been said by @theotherjimmy and @bridadan offline, the tool appears to be applying fixes to Python code that is not Py2 compatible.

That being said, looking at the files that were modified would be good to see, since that could point to either unmaintained tools, or legitimate issues.

On the topic of dead/unmaintained toolage, @theotherjimmy @brianesquilona @SenRamakri do we have a task coming up at any point in time to clean out the cobwebs in regards to tools/scripts that are no longer used?

theotherjimmy, 2019-03-28:
@cmonr  see https://github.com/ARMmbed/mbed-os/pull/10254

closed 2019-12-10 by ciarmcom