ARMmbed/mbed-os #10250: Tools: Multiple exports are failing for blinky

OPpuolitaival, 2019-03-28:
### Description

Multiple exports are failing for blinky example. 

This is how to get known which tools are supported
```mbed export -S```

Used command example:
```mbed export -m KL26Z -i uvision6```

Source link in internal jenkins: https://mbed-os.mbedcloudtesting.com/job/exporter-all-python-nightly/

These are failing

| Target | exporter |
| ------------- | ------------- |
| SAMG55J19 | uvision6 |
| VBLUNO51_BOOT | make_armc6 |
| VBLUNO51_BOOT | uvision6 |
| VBLUNO51_LEGACY | make_armc6 |
| VBLUNO51_LEGACY | uvision6 |
| VBLUNO51_OTA | make_armc6 |
| VBLUNO51_OTA | uvision6 |
| ARCH_PRO | mcuxpresso |
| LPC1768 | mcuxpresso |
| LPC1769 | mcuxpresso |
| LPC4088 | mcuxpresso |
| LPC4088_DM | mcuxpresso |
| LPC4330_M4 | mcuxpresso |
| LPCCAPPUCCINO | mcuxpresso |
| NUCLEO_L4R5ZI_P | sw4stm32 |
| UBLOX_C027 | mcuxpresso |
| VBLUNO51_BOOT | cmake_gcc_arm |
| VBLUNO51_BOOT | e2studio |
| VBLUNO51_BOOT | eclipse_gcc_arm |
| VBLUNO51_BOOT | gnuarmeclipse |
| VBLUNO51_BOOT | make_gcc_arm |
| VBLUNO51_BOOT | netbeans |
| VBLUNO51_BOOT | qtcreator |
| VBLUNO51_BOOT | vscode_gcc_arm |
| VBLUNO51_LEGACY | cmake_gcc_arm |
| VBLUNO51_LEGACY | e2studio |
| VBLUNO51_LEGACY | eclipse_gcc_arm |
| VBLUNO51_LEGACY | gnuarmeclipse |
| VBLUNO51_LEGACY | make_gcc_arm |
| VBLUNO51_LEGACY | netbeans |
| VBLUNO51_LEGACY | qtcreator |
| VBLUNO51_LEGACY | vscode_gcc_arm |
| VBLUNO51_OTA | cmake_gcc_arm |
| VBLUNO51_OTA | e2studio |
| VBLUNO51_OTA | eclipse_gcc_arm |
| VBLUNO51_OTA | gnuarmeclipse |
| VBLUNO51_OTA | make_gcc_arm |
| VBLUNO51_OTA | netbeans |
| VBLUNO51_OTA | qtcreator |
| VBLUNO51_OTA | vscode_gcc_arm |

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-04-02:
Can you please review the above failures:

@ARMmbed/team-st-mcd 
@ARMmbed/team-nxp 
@ARMmbed/team-ublox

0xc0170, 2019-04-03:
Still few more to fix!

mudassar-ublox, 2019-04-03:
`UBLOX_C027`: Not getting export error for `mcuxpresso `on blinky example.

D:\Projects\NO_MBED\mbed-os-example-blinky>mbed export -m UBLOX_C027 -i mcuxpresso
[mbed] Working path "D:\Projects\NO_MBED\mbed-os-example-blinky" (program)
Scan: mbed-os-example-blinky
Done. Import the 'mbed-os-example-blinky' project in MCUXpresso.

0xc0170, 2019-06-07:
@ARMmbed/mbed-os-test We should update this with recent data and take actions to get this fixed

OPpuolitaival, 2019-06-11:
Now there are failing these:

| Target | exporter |
| ------------- | ------------- |
|KL26Z|uvision6|
|SAMG55J19|uvision6|
|VBLUNO51_BOOT|make_armc6|
|VBLUNO51_LEGACY|make_armc6|
|VBLUNO51_OTA|make_armc6|
|ARCH_PRO|mcuxpresso|
|LPC1768|mcuxpresso|
|LPC1769|mcuxpresso|
|LPC4088|mcuxpresso|
|LPC4088_DM|mcuxpresso|
|LPC4330_M4|mcuxpresso|
|LPCCAPPUCCINO|mcuxpresso|
|UBLOX_C027|mcuxpresso|
|VBLUNO51_BOOT|cmake_gcc_arm|
|VBLUNO51_BOOT|e2studio|
|VBLUNO51_BOOT|eclipse_gcc_arm|
|VBLUNO51_BOOT|gnuarmeclipse|
|VBLUNO51_BOOT|make_gcc_arm|
|VBLUNO51_BOOT|netbeans|
|VBLUNO51_BOOT|qtcreator|
|VBLUNO51_BOOT|vscode_gcc_arm|
|VBLUNO51_LEGACY|cmake_gcc_arm|
|VBLUNO51_LEGACY|e2studio|
|VBLUNO51_LEGACY|eclipse_gcc_arm|
|VBLUNO51_LEGACY|gnuarmeclipse|
|VBLUNO51_LEGACY|make_gcc_arm|
|VBLUNO51_LEGACY|netbeans|
|VBLUNO51_LEGACY|qtcreator|
|VBLUNO51_LEGACY|vscode_gcc_arm|
|VBLUNO51_OTA|cmake_gcc_arm|
|VBLUNO51_OTA|e2studio|
|VBLUNO51_OTA|eclipse_gcc_arm|
|VBLUNO51_OTA|gnuarmeclipse|
|VBLUNO51_OTA|make_gcc_arm|
|VBLUNO51_OTA|netbeans|
|VBLUNO51_OTA|qtcreator|
|VBLUNO51_OTA|vscode_gcc_arm|

PR #10294 (2019-04-03): NUCLEO_L4R5ZI_P: enable sw4stm32 export
### Description

This partially fix #10250 

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change

### Reviewers

@OPpuolitaival
closed 2020-07-16 by ciarmcom