ARMmbed/mbed-os #10252: Unittests Disable pthreads

DBS06, 2019-03-28:
### Description

### Issue request type

Hi,

I want to compile my unittests without pthreads (gtest_disable_pthreads=ON), but I dont find the file where I need to add this flag and with the CLI its not possible to attach flags.
So my question is, how do I force unittests to build with the "gtest_disable_pthreads=ON" flag?

kR Philipp

    [X] Question
    [ ] Enhancement
    [ ] Bug

cmonr, 2019-03-28:
@ARMmbed/mbed-os-test-team ?

DBS06, 2019-04-08:
Any updates on this issue?

alekla01, 2019-04-09:
I have neither developed nor really used the Mbed OS unittests tools, but it seems python file modifications (such as https://github.com/ARMmbed/mbed-os/blob/master/UNITTESTS/unit_test/test.py#L83) may be required.

lorjala, 2019-04-09:
The CLI tool does not support this currently, but it is possible to run CMake manually with the pthread disable flag on. Follow the steps [here](https://github.com/ARMmbed/mbed-os/tree/master/UNITTESTS#build-tests-directly-with-cmake), but add an additional parameter _-Dgtest_disable_pthreads=ON_ to disable pthreads.

closed 2019-04-18 by ciarmcom

linlingao, 2019-04-30:
Please reopen this if the workaround suggested doesn't work.