ARMmbed/mbed-os #10261: LPC55S69 RTC test fail with GCC compiler

maclobdell, 2019-03-28:
### Description

RTC test failure with GCC compiler.  It passes with ARMC6 compiler.

  the RTC Glitch test case is failing.  

Board: LPCXpresso55S69

Mbed OS Version: 5.12.0

Compiler Version: GNU Tools for ARM Embedded Processors 6-2017-q1-update) 6.3.1 20170215 (release)

Test Command: mbed test -m LPC55S69_NS -t GCC_ARM

Test: tests-mbed_hal-rtc x
Result: FAIL

Platform: LPC55S69_NS - Toolchain: GCC_ARM
Elapsed Time: 27.52 seconds
Build Path: BUILD/tests/LPC55S69_NS/GCC_ARM
Absolute Build Path: C:\Users\maclob01\Documents\mbed-os-5\mbed-os-cy-nxp\BUILD\tests\LPC55S69_NS\GCC_ARM
Copy Method: default
Image Path: BUILD/tests/LPC55S69_NS/GCC_ARM/TESTS/mbed_hal/rtc/rtc.hex

See attached test log
[lpc55s69_evk-rtc-test-fail.txt](https://github.com/ARMmbed/mbed-os/files/3019651/lpc55s69_evk-rtc-test-fail.txt)

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

cc @mmahadevan108

maclobdell, 2019-04-30:
I tested with GCC compiler version 7.3.1 and it works.  

The guidance I have received is that Arm v8m targets require GCC 7.  

Closing this ticket.

maclobdell, 2019-05-07:
re-opening, because we need better root cause of why it is failing with GCC 6.  Mbed OS is not planning to update to GCC 7 in 5.13, so we need to see if GCC 6 can work for the next several months.

mmahadevan108, 2019-05-07:
I am unable to reproduce the failure with GCC 6.

maclobdell, 2019-05-08:
I just retested with GCC 6 and it passed.

closed 2019-05-09 by ciarmcom