ARMmbed/mbed-os #10262: LPC55S69 KVSTORE test fails with GCC compiler

maclobdell, 2019-03-28:
### Description

Board: LPCXpresso55S69

Mbed OS Version: 5.12.0

Compiler Version: GNU Tools for ARM Embedded Processors 6-2017-q1-update) 6.3.1 20170215 (release)

Test Command: mbed test -m LPC55S69_NS -t GCC_ARM

Test: features-storage-tests-kvstore-static_tests x
Result: TIMEOUT

Platform: LPC55S69_NS - Toolchain: GCC_ARM
Elapsed Time: 3021.98 seconds
Build Path: BUILD/tests/LPC55S69_NS/GCC_ARM
Absolute Build Path: C:\Users\maclob01\Documents\mbed-os-5\mbed-os-cy-nxp\BUILD\tests\LPC55S69_NS\GCC_ARM
Copy Method: default
Image Path: BUILD/tests/LPC55S69_NS/GCC_ARM/features/storage/TESTS/kvstore/static_tests/static_tests.hex

This test case fails.  Many other test cases are then skipped.

Testcase: set_several_keys_multithreaded

Result: ERROR
Elapsed Time: 0.00
Start Time: 28-03-2019 00:25:02.160000
End Time: 28-03-2019 00:25:02.160000
Failed: 0
Passed: 0

uTest Log

[1553750702.11][CONN][RXD] >>> Running case #7: 'set_several_keys_multithreaded'...
[1553750702.16][CONN][INF] found KV pair in stream: {{__testcase_start;set_several_keys_multithreaded}}, queued...
[1553750702.22][CONN][RXD] :140::FAIL: Expected 0 Was 284
[1553750702.22][CONN][RXD] :0::PASS
[1553753698.36][HTST][INF] test suite run finished after 3000.06 sec...
[1553753698.36][CONN][INF] received special event '__host_test_finished' value='True', finishing
[1553753698.41][HTST][INF] CONN exited with code: 0
[1553753698.41][HTST][INF] No events in queue
[1553753698.41][HTST][INF] stopped consuming events
[1553753698.41][HTST][INF] host test result(): None
[1553753698.41][HTST][WRN] missing __exit event from DUT
[1553753698.41][HTST][WRN] missing __exit_event_queue event from host test
[1553753698.41][HTST][ERR] missing __exit_event_queue event from host test and no result from host test, timeout...
[1553753698.41][HTST][INF] calling blocking teardown()
[1553753698.41][HTST][INF] teardown() finished
[1553753698.41][HTST][INF] {{result;timeout}}

Full Test Log
[lpc55s69_evk-kvstore-test-fail.txt](https://github.com/ARMmbed/mbed-os/files/3019676/lpc55s69_evk-kvstore-test-fail.txt)

cc @mmahadevan108 

### Issue request type
    [ ] Question
    [ ] Enhancement
    [ ] Bug

mmahadevan108, 2019-03-28:
Please try with the below GCC toolchain.

arm-none-eabi-gcc.exe (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release)

mmahadevan108, 2019-03-29:
I don't see the failure. Below is the test run results with IAR. With GCC, the toolchain should be updated.
mbedgt: test suite 'mbed-os-features-storage-tests-kvstore-static_tests' ............................. OK in 40.17 sec
        test case: 'get_buffer_null_size_not_zero' ................................................... OK in 0.06 sec
        test case: 'get_buffer_size_bigger_than_data_real_size' ...................................... OK in 0.07 sec
        test case: 'get_buffer_size_is_zero' ......................................................... OK in 0.05 sec
        test case: 'get_buffer_size_smaller_than_data_real_size' ..................................... OK in 0.07 sec
        test case: 'get_info_existed_key' ............................................................ OK in 0.06 sec
        test case: 'get_info_info_null' .............................................................. OK in 0.04 sec
        test case: 'get_info_key_length_exceeds_max' ................................................. OK in 0.06 sec
        test case: 'get_info_key_null' ............................................................... OK in 0.04 sec
        test case: 'get_info_non_existing_key' ....................................................... OK in 0.05 sec
        test case: 'get_info_overwritten_key' ........................................................ OK in 0.07 sec
        test case: 'get_info_removed_key' ............................................................ OK in 0.05 sec
        test case: 'get_key_length_exceeds_max' ...................................................... OK in 0.06 sec
        test case: 'get_key_null' .................................................................... OK in 0.05 sec
        test case: 'get_key_that_was_set_twice' ...................................................... OK in 0.07 sec
        test case: 'get_non_existing_key' ............................................................ OK in 0.04 sec
        test case: 'get_removed_key' ................................................................. OK in 0.04 sec
        test case: 'get_several_keys_multithreaded' .................................................. OK in 0.09 sec
        test case: 'iterator_close_right_after_iterator_open' ........................................ OK in 0.08 sec
        test case: 'iterator_next_empty_list' ........................................................ OK in 0.05 sec
        test case: 'iterator_next_empty_list_keys_removed' ........................................... OK in 0.07 sec
        test case: 'iterator_next_empty_list_non_matching_prefix' .................................... OK in 0.08 sec
        test case: 'iterator_next_full_list' ......................................................... OK in 0.08 sec
        test case: 'iterator_next_key_size_zero' ..................................................... OK in 0.07 sec
        test case: 'iterator_next_one_key_list' ...................................................... OK in 0.05 sec
        test case: 'iterator_next_path_check' ........................................................ OK in 0.06 sec
        test case: 'iterator_next_remove_while_iterating' ............................................ OK in 0.13 sec
        test case: 'iterator_next_several_overwritten_keys' .......................................... OK in 0.10 sec
        test case: 'iterator_open_it_null' ........................................................... OK in 0.05 sec
        test case: 'kvstore_init' .................................................................... OK in 0.05 sec
        test case: 'remove_existed_key' .............................................................. OK in 0.04 sec
        test case: 'remove_key_length_exceeds_max' ................................................... OK in 0.07 sec
        test case: 'remove_key_null' ................................................................. OK in 0.05 sec
        test case: 'remove_non_existing_key' ......................................................... OK in 0.05 sec
        test case: 'remove_removed_key' .............................................................. OK in 0.05 sec
        test case: 'set_buffer_null_size_not_zero' ................................................... OK in 0.06 sec
        test case: 'set_buffer_size_is_zero' ......................................................... OK in 0.04 sec
        test case: 'set_key_length_exceeds_max' ...................................................... OK in 0.05 sec
        test case: 'set_key_null' .................................................................... OK in 0.03 sec
        test case: 'set_key_value_fifteen_byte_size' ................................................. OK in 0.06 sec
        test case: 'set_key_value_five_byte_size' .................................................... OK in 0.06 sec
        test case: 'set_key_value_one_byte_size' ..................................................... OK in 0.07 sec
        test case: 'set_key_value_seventeen_byte_size' ............................................... OK in 0.06 sec
        test case: 'set_key_value_two_byte_size' ..................................................... OK in 0.07 sec
        test case: 'set_same_key_several_time' ....................................................... OK in 0.06 sec
[... cut ...]
               | 1      | 0      | OK     | 0.06               |
| LPC55S69_NS-IAR | LPC55S69      | mbed-os-features-storage-tests-kvstore-static_tests | set_write_once_flag_try_remove
               | 1      | 0      | OK     | 0.06               |
| LPC55S69_NS-IAR | LPC55S69      | mbed-os-features-storage-tests-kvstore-static_tests | set_write_once_flag_try_set_tw
ice            | 1      | 0      | OK     | 0.06               |
mbedgt: test case results: 49 OK
mbedgt: completed in 69.00 sec

mmahadevan108, 2019-03-29:
The above was run on mbed-os master

mmahadevan108, 2019-03-29:
I tried with tag mbed-os-5.12.0 and the tests passed as well.

mmahadevan108, 2019-03-29:
@dannybenor . Please give it a try as well.

maclobdell, 2019-03-29:
Thanks a lot for confirming this is working on your side Mahesh.

Our documentation currently states that the supported GCC compiler is 
GNU Arm Embedded version 6 (6-2017-q1-update).

https://os.mbed.com/docs/mbed-os/v5.12/tools/index.html

I'll check if this is correct, or needs to be reviewed/updated.

maclobdell, 2019-03-29:
confirmed with our team that version 6 is still the officially supported version.  

@mmahadevan108 - can you investigate further with gcc 6 and see if there is a possible workaround?

@dannybenor  - since this is kvstore related, can you confirm which compiler version you have been using, and if you have seen this issue?

mmahadevan108, 2019-03-29:
@maclobdell Cortex-M33 is ARMv8M architecture which is supported by ARMC6 toolchain, GCC ARM version 7 (higher) and IAR 8.x.  Hence you would need to upgrade your GCC toolchain.

0xc0170, 2019-04-02:
The compiler versions are correct however . ARMv8M is not supported in GCC ARM 6. @ARMmbed/mbed-os-psa is that correct?

orenc17, 2019-04-02:
@0xc0170 GCC_ARM 6 is the first version of GCC that supports V8m

0xc0170, 2019-04-02:
> @0xc0170 GCC_ARM 6 is the first version of GCC that supports V8m

@mmahadevan108 This is a valid bug in the currently supported GCC ARM version

mmahadevan108, 2019-04-02:
@orenc17 @dannybenor Please confirm if this is a valid issue, I am unable to reproduce.

orenc17, 2019-04-02:
i've run the test on master with all 3 toolchains with release and debug profile
and they seem to pass

https://mbed-os.mbedcloudtesting.com/blue/organizations/jenkins/PSA-SPM-tests/detail/PSA-SPM-tests/40/pipeline/

cmonr, 2019-04-08:
@maclobdell Is this still an issue? Multiple people cannot reproduce.

maclobdell, 2019-04-12:
I just tested Mbed OS 5.12.1 with GCC compiler 7.3.1 

running features-storage-tests-kvstore-static_tests results in the below results.

could the hardware I'm testing have a problem?  Or maybe some expected configuration of the chip is not right?  Or maybe DAPLink version has a problem?  

Can someone help diagnose the test results? 

```
        test filtered in 'features-storage-tests-kvstore-static_tests'
mbedgt: running 1 test for platform 'LPC55S69' and toolchain 'GCC_ARM'
mbedgt: mbed-host-test-runner: started
mbedgt: :297::FAIL: Expected 0 Was 284
mbedgt: :297::FAIL: Expected 0 Was 284
mbedgt: :297::FAIL: Expected 0 Was 284
mbedgt: :309::FAIL: Expected 0 Was 284
mbedgt: :309::FAIL: Expected 0 Was 284
mbedgt: :309::FAIL: Expected 0 Was 284
mbedgt: :309::FAIL: Expected 0 Was 284
mbedgt: :309::FAIL: Expected 0 Was 284
mbedgt: :309::FAIL: Expected 0 Was 284
mbedgt: :309::FAIL: Expected 0 Was 284
mbedgt: :309::FAIL: Expected 0 Was 284
mbedgt: :309::FAIL: Expected 0 Was 284
mbedgt: :309::FAIL: Expected 0 Was 284
mbedgt: :309::FAIL: Expected 0 Was 284
mbedgt: :309::FAIL: Expected 0 Was 284
mbedgt: :309::FAIL: Expected 0 Was 284
mbedgt: :309::FAIL: Expected 0 Was 284
mbedgt: :304::FAIL: Expected 0 Was 263
mbedgt: :305::FAIL: Expected 'name_v' Was ''
mbedgt: :309::FAIL: Expected 0 Was 263
mbedgt: :309::FAIL: Expected 0 Was 284
mbedgt: :304::FAIL: Expected 0 Was 263
mbedgt: :305::FAIL: Expected 'name_x' Was ''
mbedgt: :309::FAIL: Expected 0 Was 263
mbedgt: :304::FAIL: Expected 0 Was 263
mbedgt: :305::FAIL: Expected 'name_y' Was ''
mbedgt: :309::FAIL: Expected 0 Was 263
mbedgt: :309::FAIL: Expected 0 Was 284
mbedgt: :373::FAIL: Expected 0 Was 284
mbedgt: :383::FAIL: Expected 0 Was 284
mbedgt: :387::FAIL: Expected '' Was 'data'
mbedgt: :399::FAIL: Expected 0 Was 284
mbedgt: :403::FAIL: Expected 0 Was 263
mbedgt: :404::FAIL: Expected '' Was 'data'
mbedgt: :408::FAIL: Expected 0 Was 263
mbedgt: :424::FAIL: Expected 0 Was 284
mbedgt: :427::FAIL: Expected 0 Was 263
mbedgt: :442::FAIL: Expected 0 Was 284
mbedgt: :446::FAIL: Expected 'data' Was 'new_data'
mbedgt: :473::FAIL: Expected 0 Was 284
mbedgt: :473::FAIL: Expected 0 Was 284
mbedgt: :459::FAIL: :459:Expec:FAILted 0: Exp Wasected 263
mbedgt: :489::FAIL: Expected 0 Was 263
mbedgt: :489::FAIL: Expected 0 Was 263
mbedgt: :489::FAIL: Expected 0 Was 284
mbedgt: :527::FAIL: Expected 0 Was 284
mbedgt: :530::FAIL: Expected 0 Was 263
mbedgt: :544::FAIL: Expected 0 Was 284
```

cc @ARMmbed/mbed-os-storage

mmahadevan108, 2019-04-12:
Can you try a different board. Also I am using the DAPLink binaries from the below link:
https://os.mbed.com/teams/NXP/wiki/Updating-LPCXpresso-firmware

cmonr, 2019-04-16:
@ARMmbed/mbed-os-storage

alzix, 2019-04-16:
@cmonr, it is not reproduced on our side

0xc0170, 2019-04-18:
@maclobdell closing or talk to @alzix

maclobdell, 2019-04-24:
I've tried multiple boards.  I've tried updating the Daplink to the latest.  I get consistent failures.  Even if I go back to ARM compiler 6, the tests eventually time out.  There must be some misunderstood dependency.  The tests are not stable.  Can I get some guidance and help here?

yossi2le, 2019-04-28:
Hi @maclobdell, can you please provide your KVStore configuration under features/storage/kvstore/conf?

maclobdell, 2019-04-29:
I'm testing LPC55S69 with Mbed OS 5.12.1 with no changes in features/storage/kvstore/conf

orenc17, 2019-04-30:
I've run the `mbed-os-features-storage-tests-kvstore-static_tests` test on all 3 toolchains using both debug and release profiles and they all PASS

https://mbed-os.mbedcloudtesting.com/view/PSA/job/PSA-SPM-tests/54/ - 5.12.1 GCC & IAR
https://mbed-os.mbedcloudtesting.com/view/PSA/job/PSA-SPM-tests/56/ - 5.12.1 ARMC6
https://mbed-os.mbedcloudtesting.com/view/PSA/job/PSA-SPM-tests/57/ - 5.12.2 ARMC6, GCC & IAR

@ARMmbed/mbed-os-maintainers i believe we can close the issue

maclobdell, 2019-05-01:
Based on suggestion from Mahesh at NXP, I set jumper J10, erased the flash using Keil uVision, removed J10, then used a freshly installed Mbed OS 5.12.2, I ran the tests with GCC 7.3.1 and the tests all passed.

maclobdell, 2019-05-01:
so there is definitely a dependency on the initial state of flash.

maclobdell, 2019-05-01:
also, I noticed that if I switch from GCC_ARM to IAR, I have to erase the flash in-between, otherwise it will fail with IAR.

flit, 2019-06-19:
You should *never* have to manually wipe the flash in order to get `TDBStore` to work.

After digging a little into this, it's clear there is a bug in the LPC55S69 target support that causes `flash_read()` to default to the version in `hal/mbed_flash_api.c`. Instead, it should be using the NS veneer (from the cmse lib) for `flash_read()` from `.../TARGET_LPC55S69/flash_api.c`, thus making a call into the secure world version of this API.

It seems like there might be some conflict caused by the default `flash_read()` being weak?