ARMmbed/mbed-os #10264: LPC55S69 Storage Test Failure with IAR 8.32

maclobdell, 2019-03-28:
### Description

Board: LPCXpresso55S69

Mbed OS Version: 5.12.0

Compiler: IAR 8.32

Test Command: mbed test -m LPC55S69_NS -t GCC_ARM

Test: features-storage-tests-filesystem-general_filesystem 

Result: SYNC_FAILED
Platform: LPC55S69_NS - Toolchain: IAR Elapsed Time: 45.41 seconds Build Path: BUILD/tests/LPC55S69_NS/IAR Absolute Build Path: D:\mbed\mbed-os\BUILD\tests\LPC55S69_NS\IAR Copy Method: default Image Path: BUILD/tests/LPC55S69_NS/IAR/features/storage/TESTS/filesystem/general_filesystem/general_filesystem.hex

Test Log
[lpc55s69_evk-kvstore-test-fail.txt](https://github.com/ARMmbed/mbed-os/files/3019930/lpc55s69_evk-kvstore-test-fail.txt)

cc @mmahadevan108 
cc @dannybenor 

### Issue request type
    [ ] Question
    [ ] Enhancement
    [x] Bug

maclobdell, 2019-04-12:
any update on this?  I tested Mbed OS v5.12.1 and get the same result.  It passes with ARMC6 and fails with IAR.  Other tests pass with IAR.

I have IAR compiler version: 
IAR ANSI C/C++ Compiler V8.32.3.193/W32 for ARM

this is the latest version downloaded from IAR.  

the tools documentation states that V8.32.1 is the supported version by Mbed OS.  could this issue be a result of using a slightly newer patch release of the compiler?  seems unlikely to me but I'm not sure.

please help!  this is blocking Mbed Enabled for this board.

cmonr, 2019-04-12:
@ARMmbed/mbed-os-storage 

@maclobdell Maybe it would help to post/attach the full failure log?

maclobdell, 2019-04-12:
additional log attached.
[lpc55s69-storage-test-fail.txt](https://github.com/ARMmbed/mbed-os/files/3071407/lpc55s69-storage-test-fail.txt)

maclobdell, 2019-05-01:
This is still an issue.  @ARMmbed/mbed-os-storage  or @mmahadevan108 any advice?

I re-ran the test with a fresh mbed os version and verbose output.  

I can see UnicodeDecodeError and bad characters coming through.  I have no idea why this test would behave this way, but others do not.  Also the failures only occur with IAR.    

```
[1556743510.58][HTST][INF] host test executor ver. 0.0.2
[1556743510.58][HTST][INF] copy image onto target...
[1556743510.58][COPY][INF] Waiting up to 60 sec for '02360000028602ba00000000000000000000000097969905' mount point (current is 'G:')...
        1 file(s) copied.
[1556743534.10][HTST][INF] starting host test process...
[1556743534.50][CONN][INF] starting connection process...
[1556743534.50][CONN][INF] notify event queue about extra 60 sec timeout for serial port pooling
[1556743534.50][CONN][INF] initializing serial port listener...
[1556743534.50][PLGN][INF] Waiting up to 60 sec for '02360000028602ba00000000000000000000000097969905' serial port (current is 'COM9')...
[1556743534.51][HTST][INF] setting timeout to: 60 sec
[1556743534.55][SERI][INF] serial(port=COM9, baudrate=9600, read_timeout=0.01, write_timeout=5)
[1556743534.55][SERI][INF] reset device using 'default' plugin...
[1556743534.80][SERI][INF] waiting 1.00 sec after reset
[1556743535.80][SERI][INF] wait for it...
[1556743535.80][SERI][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
[1556743535.80][CONN][INF] sending up to 5 __sync packets (specified with --sync=5)
[1556743535.81][CONN][INF] sending preamble 'c90d42df-6ac1-4658-8de9-56ea5476196a'
[1556743535.81][SERI][TXD] {{__sync;c90d42df-6ac1-4658-8de9-56ea5476196a}}
[1556743535.82][CONN][WRN] UnicodeDecodeError encountered!
[1556743535.82][CONN][RXD]             ☺    lfs error:494: Corrupted dir pair at 0 1
[1556743535.82][CONN][RXD] lfs error:2216: Invalid superblock at 0 1
[1556743540.81][CONN][INF] Reset the part and send in new preamble...
[1556743540.81][SERI][INF] reset device using 'default' plugin...
[1556743541.06][SERI][INF] waiting 1.00 sec after reset
[1556743542.06][SERI][INF] wait for it...
[1556743542.06][CONN][INF] resending new preamble 'c17eaa28-9c9d-4c61-9c2d-91d0dd6532a4' after 5.00 sec
[1556743542.06][SERI][TXD] {{__sync;c17eaa28-9c9d-4c61-9c2d-91d0dd6532a4}}
[1556743542.07][CONN][WRN] UnicodeDecodeError encountered!
[1556743542.07][CONN][RXD]                    lfs error:494: Corrupted dir pair at 0 1
[1556743542.07][CONN][RXD] lfs error:2216: Invalid superblock at 0 1
[1556743547.06][CONN][INF] Reset the part and send in new preamble...
[1556743547.06][SERI][INF] reset device using 'default' plugin...
[1556743547.31][SERI][INF] waiting 1.00 sec after reset
[1556743548.31][SERI][INF] wait for it...
[1556743548.31][CONN][INF] resending new preamble 'eda267a1-e96a-4e47-91f6-b716461fd725' after 5.00 sec
[1556743548.31][SERI][TXD] {{__sync;eda267a1-e96a-4e47-91f6-b716461fd725}}
[1556743548.32][CONN][WRN] UnicodeDecodeError encountered!
[1556743548.33][CONN][RXD]                    lfs error:494: Corrupted dir pair at 0 1
[1556743548.33][CONN][RXD] lfs error:2216: Invalid superblock at 0 1
[1556743553.32][CONN][INF] Reset the part and send in new preamble...
[1556743553.32][SERI][INF] reset device using 'default' plugin...
[1556743553.57][SERI][INF] waiting 1.00 sec after reset
[1556743554.57][SERI][INF] wait for it...
[1556743554.57][CONN][INF] resending new preamble '39c57fc0-bed1-4b66-a455-aff1d0c571e9' after 5.00 sec
[1556743554.57][SERI][TXD] {{__sync;39c57fc0-bed1-4b66-a455-aff1d0c571e9}}
[1556743554.58][CONN][WRN] UnicodeDecodeError encountered!
[1556743554.58][CONN][RXD]    ∟  ∟  8               lfs error:494: Corrupted dir pair at 0 1
[1556743554.58][CONN][RXD] lfs error:2216: Invalid superblock at 0 1
[1556743559.57][CONN][INF] Reset the part and send in new preamble...
[1556743559.57][SERI][INF] reset device using 'default' plugin...
[1556743559.82][SERI][INF] waiting 1.00 sec after reset
[1556743560.82][SERI][INF] wait for it...
[1556743560.82][CONN][INF] resending new preamble 'b1e960cd-a948-4ce0-b970-06e53c2f48f9' after 5.00 sec
[1556743560.83][SERI][TXD] {{__sync;b1e960cd-a948-4ce0-b970-06e53c2f48f9}}
[1556743560.84][CONN][WRN] UnicodeDecodeError encountered!
[1556743560.84][CONN][RXD]        ∟   `             lfs error:494: Corrupted dir pair at 0 1
[1556743560.84][CONN][RXD] lfs error:2216: Invalid superblock at 0 1
[1556743560.86][HTST][ERR] None
[1556743560.86][HTST][WRN] stopped to consume events due to __notify_sync_failed event
[1556743560.86][HTST][INF] __exit_event_queue received
[1556743560.86][HTST][INF] test suite run finished after 26.35 sec...
[1556743560.88][HTST][INF] CONN exited with code: 0
[1556743560.88][HTST][INF] No events in queue
[1556743560.88][HTST][INF] host test result() call skipped, received: sync_failed
[1556743560.88][HTST][WRN] missing __exit event from DUT
[1556743560.88][HTST][INF] calling blocking teardown()
[1556743560.88][HTST][INF] teardown() finished
[1556743560.88][HTST][INF] {{result;sync_failed}}
```

yossi2le, 2019-05-02:
@maclobdell, actually the general_filesystem tests are not supported for FLASHIAP. If you look at the code you will see that all tests are skipped. The problem is that the initialization is static and therefore fails under IAR. I have fixed it and instead of using skipped it will now report as not supported. 
Thanks for bringing this to our attention. 
The fix can be found at PR #10523

maclobdell, 2019-05-07:
@yossi2le - thanks very much!

michalpasztamobica, 2019-08-12:
@maclobdell , @cmonr - anything else is required here? Can we close?

0xc0170, 2019-08-12:
> The fix can be found at PR #10523

Should be fixed

closed 2019-08-12 by 0xc0170