ARMmbed/mbed-os #10266: mbed_hal-sleep test fails when run with debug profile.

sathysreenath, 2019-03-28:
### Description
When compiled with debug profile (no optimization) mbed-os-tests-mbed_hal-sleep | deep-sleep - high-speed clocks are turned off, fails. We are using K64F board and IAR (8.30) toolchain .

We tried this on the latest master mbed-os and mbed-cli (1.10.0). 

Here are the command lines we are using for the mbed compiler:
mbed test --source mbed-os --compile --clean  --target=K64F --profile mbed-os/tools/profiles/debug.json -n mbed-os-tests-mbed_hal-sleep

mbed test --run --copy jlink  -r jlink --target=K64F -vv --profile mbed-os/tools/profiles/debug.json -n  mbed-os-tests-mbed_hal-sleep

( or use htrun directly: mbedhtrun -d D: -p com12 -f sleep.bin -m 64F)

Here is the output:

[1553803927.53][HTST][INF] host test executor ver. 1.4.1
[1553803927.53][HTST][INF] copy image onto target...
        1 file(s) copied.
[1553803934.90][MBED][WRN] Target ID not found: Skipping flash check and retry
[1553803934.95][HTST][INF] starting host test process...
[1553803935.70][CONN][INF] starting connection process...
[1553803935.70][CONN][INF] notify event queue about extra 60 sec timeout for serial port pooling
[1553803935.71][CONN][INF] initializing serial port listener...
[1553803935.71][SERI][INF] serial(port=COM12, baudrate=9600, read_timeout=0.01, write_timeout=5)
[1553803935.71][SERI][INF] reset device using 'default' plugin...
[1553803935.71][HTST][INF] setting timeout to: 60 sec
[1553803935.96][SERI][INF] waiting 1.00 sec after reset
[1553803936.96][SERI][INF] wait for it...
[1553803936.96][SERI][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
[1553803936.96][CONN][INF] sending up to 2 __sync packets (specified with --sync=2)
[1553803936.96][CONN][INF] sending preamble '2921b727-cd45-4b38-855f-8aa845e1efa6'
[1553803936.96][SERI][TXD] {{__sync;2921b727-cd45-4b38-855f-8aa845e1efa6}}
[1553803937.09][CONN][RXD] mbedmbedmbedmbedmbedmbedmbedmbed
[1553803937.14][CONN][INF] found SYNC in stream: {{__sync;2921b727-cd45-4b38-855f-8aa845e1efa6}} it is #0 sent, queued...
[1553803937.15][HTST][INF] sync KV found, uuid=2921b727-cd45-4b38-855f-8aa845e1efa6, timestamp=1553803937.142000
[1553803937.16][CONN][INF] found KV pair in stream: {{__version;1.3.0}}, queued...
[1553803937.17][HTST][INF] DUT greentea-client version: 1.3.0
[1553803937.19][CONN][INF] found KV pair in stream: {{__timeout;60}}, queued...
[1553803937.19][HTST][INF] setting timeout to: 60 sec
[1553803937.22][CONN][INF] found KV pair in stream: {{__host_test_name;default_auto}}, queued...
[1553803937.22][HTST][INF] host test class: '<class 'mbed_host_tests.host_tests.default_auto.DefaultAuto'>'
[1553803937.22][HTST][INF] host test setup() call...
[1553803937.22][HTST][INF] CALLBACKs updated
[1553803937.22][HTST][INF] host test detected: default_auto
[1553803937.24][CONN][INF] found KV pair in stream: {{__testcase_count;3}}, queued...
[1553803937.30][CONN][RXD] >>> Running 3 test cases...
[1553803937.34][CONN][INF] found KV pair in stream: {{__testcase_name;sleep - source of wake-up - us ticker}}, queued...
[1553803937.41][CONN][INF] found KV pair in stream: {{__testcase_name;deep-sleep - source of wake-up - lp ticker}}, queued...
[1553803937.47][CONN][RXD]
[1553803937.47][CONN][INF] found KV pair in stream: {{__testcase_name;deep-sleep - high-speed clocks are turned off}}, queued...
[1553803937.54][CONN][RXD] >>> Running case #1: 'sleep - source of wake-up - us ticker'...
[1553803937.60][CONN][INF] found KV pair in stream: {{__testcase_start;sleep - source of wake-up - us ticker}}, queued...
[1553803937.70][CONN][INF] found KV pair in stream: {{__testcase_finish;sleep - source of wake-up - us ticker;1;0}}, queued...
[1553803937.76][CONN][RXD] >>> 'sleep - source of wake-up - us ticker': 1 passed, 0 failed
[1553803937.76][CONN][RXD]
[1553803937.84][CONN][RXD] >>> Running case #2: 'deep-sleep - source of wake-up - lp ticker'...
[1553803937.90][CONN][INF] found KV pair in stream: {{__testcase_start;deep-sleep - source of wake-up - lp ticker}}, queued...
[1553803938.89][CONN][INF] found KV pair in stream: {{__testcase_finish;deep-sleep - source of wake-up - lp ticker;1;0}}, queued...
[1553803938.96][CONN][RXD] >>> 'deep-sleep - source of wake-up - lp ticker': 1 passed, 0 failed
[1553803938.96][CONN][RXD]
[1553803939.04][CONN][RXD] >>> Running case #3: 'deep-sleep - high-speed clocks are turned off'...
[1553803939.11][CONN][INF] found KV pair in stream: {{__testcase_start;deep-sleep - high-speed clocks are turned off}}, queued...
[1553803939.21][CONN][RXD] :174::FAIL: Values Not Within Delta 1000 Expected 0 Was 19977
[1553803939.29][CONN][INF] found KV pair in stream: {{__testcase_finish;deep-sleep - high-speed clocks are turned off;0;1}}, queued...
[1553803939.39][CONN][RXD] >>> 'deep-sleep - high-speed clocks are turned off': 0 passed, 1 failed with reason 'Test Cases Failed'
[1553803939.39][CONN][RXD]
[1553803939.47][CONN][RXD] >>> Test cases: 2 passed, 1 failed with reason 'Test Cases Failed'
[1553803939.49][CONN][RXD] >>> TESTS FAILED!
[1553803939.51][CONN][INF] found KV pair in stream: {{__testcase_summary;2;1}}, queued...
[1553803939.54][CONN][INF] found KV pair in stream: {{end;failure}}, queued...
[1553803939.54][HTST][INF] __notify_complete(False)
[1553803939.54][HTST][INF] __exit_event_queue received
[1553803939.54][HTST][INF] test suite run finished after 2.35 sec...
[1553803939.55][CONN][INF] found KV pair in stream: {{__exit;0}}, queued...
[1553803939.55][CONN][INF] received special event '__host_test_finished' value='True', finishing
[1553803939.58][HTST][INF] CONN exited with code: 0
[1553803939.58][HTST][INF] Some events in queue
[1553803939.58][HTST][INF] stopped consuming events
[1553803939.58][HTST][INF] host test result() call skipped, received: False
[1553803939.58][HTST][WRN] missing __exit event from DUT
[1553803939.58][HTST][INF] calling blocking teardown()
[1553803939.58][HTST][INF] teardown() finished
[1553803939.58][HTST][INF] {{result;failure}}

Thanks.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

studavekar, 2019-03-29:
@sathysreenath deep sleep is disabled in debug profile

https://github.com/ARMmbed/mbed-os/blob/2d9c4946ca3403bd064a91acbce730b66a62190a/platform/mbed_sleep_manager.c#L214-L224

sathysreenath, 2019-04-01:
Thanks for letting us know. We will disable this test during debug mode.

Sathy.

closed 2019-04-02 by ciarmcom