ARMmbed/mbed-os #10273: TFM: assemble.py is using hardcoded values

jeromecoutant, 2019-03-29:
### Description

Due to assemble script:
https://github.com/ARMmbed/mbed-os/blob/0f6a8357e73c645cfe40affa3a2145142f8f8ac4/tools/psa/tfm/bin_utils/assemble.py#L30

In flash_layout.h partition file,
SECURE and NON_SECURE IMAGE_OFFSET and IMAGE_MAX_SIZE vales
have to be hard coded value.
They can't be defined thanks to other system macro or with formulas.

### Issue request type

    [ ] Question
    [x] Enhancement
    [ ] Bug

jeromecoutant, 2019-04-23:
Please reopen this question.... :-(

jeromecoutant, 2019-04-24:
@linlingao

0xc0170, 2019-04-30:
@ARMmbed/mbed-os-psa Please review

orenc17, 2019-04-30:
This tool was provided by TF-M, and was imported as is
we do not have plans to tweak it

alzix, 2019-04-30:
please consider contributing a patch to https://git.trustedfirmware.org/trusted-firmware-m.git/ and we will import it from there

jeromecoutant, 2019-05-24:
Please review 
https://review.trustedfirmware.org/c/trusted-firmware-m/+/1145

0xc0170, 2019-08-14:
> https://review.trustedfirmware.org/c/trusted-firmware-m/+/1145

As this was merged, is it already also on master here or waiting for the next TFM update?

closed 2019-10-31 by ciarmcom