ARMmbed/mbed-os #10284: How to port running project from i.MXRT1050 to i.MXRT1060

marcin-ch, 2019-04-01:
### Description
```
Windows 7
MBED CLI (1.8.3)
GCC_ARM (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]
MIMXRT1050_EVK
MIMXRT1060_EVK
```
I have project up and running on i.MXRT1050-EVK.
How to update the project to have it working on i.MXRT1060-EVK?
Any guides, tutorials?

I do develop an application using MBED CLI and GCC toolchain. 
@maclobdell 

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

maclobdell, 2019-04-01:
The 1060 is not officially supported, but I believe it is very possible use use the 1050 codebase on a 1060.  

@mmahadevan108  for suggested steps.

cmonr, 2019-04-01:
@ARMmbed/team-nxp

mmahadevan108, 2019-04-17:
1050 & 1060 are similar. You should be able to reuse most of 1050 support with changes to the linker files. Please download the SDK for 1060 from 
https://mcuxpresso.nxp.com/en/select

closed 2019-04-18 by ciarmcom

linlingao, 2019-04-30:
@marcin-ch It appears your question has been answered. Please reopen if you still have questions.