ARMmbed/mbed-os #10297: Since mbed-os 5.12 : Compilation error : Additional properties are not allowed ('requires' was unexpected)

nmathon, 2019-04-02:
### Description

Since mbed-os 5.12, I'm not able to compile my project anymore. I have the following error :

Building project gsboard (DISCO_F746NG, ARMC6) Scan: gsboard [ERROR] in ./mbed-os/features/nanostack/sal-stack-nanostack-eventloop/mbed_lib.json element : Additional properties are not allowed ('requires' was unexpected)

Any idea ? Everything was ok before.

I am using MbedStudio with the default bundled tools.
The target is DISCO-F746NG

### Issue request type
    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-04-04:
Is this IDE error ? What version are you using? How to reproduce the issue?

Could you try to reproduce with mbed-cli ?

nmathon, 2019-04-08:
Using mbed-cli, it works. But I'm not sure to use the same toolchain. It's not really clear what's "bundled tools" means in MbedStudio.
Mbed Studio : 0.4.0
Mbed OS : 5.12.0

I just launched my product on Mbed Studio, if I'm on Mbed 0S 5.11.0, it will compile, if i'm updating Mbed OS to 5.12.0, I have the following error:

Using targets from ./mbed-os/targets/targets.json
Building project gsboard (DISCO_F746NG, ARMC6)
Scan: gsboard
[ERROR] in ./mbed-os/features/nanostack/sal-stack-nanostack-eventloop/mbed_lib.json element : Additional properties are not allowed ('requires' was unexpected)

lritchie, 2019-04-11:
Make sure you are using mbed-os tools from active program and not the bundled tools in Mbed Studio v0.4.

nmathon, 2019-04-15:
OK thanks, but that doesn't work either because Mbed Studio is bundled with python2.7.
So I will have this kind of errors instead:
UnicodeDecodeError: 'ascii' codec can't decode byte 0xe2 in position 15737: ordinal not in range(128)

0xc0170, 2019-04-15:
cc @arekzaluski

nmathon, 2019-04-15:
OK, I was able to fix that by adding the following lines in the make.py script
```
    reload(sys)  
    sys.setdefaultencoding('utf8')
```

So It works with the embedded python 2.7 now and with the mbed-os tools from active program activated

closed 2019-04-15 by nmathon

0xc0170, 2019-04-15:
Current workaround for this issue (I can see you fixed the above in the tools directly) is to select in dropdown menu to use local version of tools. 
This will be fixed in the next version.

GregorioMarquez, 2019-07-30:
This was fixed for me by upgrading from Mbed Studio v0.4 to v0.5.3