ARMmbed/mbed-os #10298: Error in Timer/Timeout class(es).

hesolium, 2019-04-02:
As in title. Times returned by Time method 'read_high_resolution_us' in subsequent calls are not monotonic. Sometimes returned value is SMALLER than value returned previously (time loops:)
Here is short test program that shows this behavior.
[main.cpp.txt](https://github.com/ARMmbed/mbed-os/files/3035131/main.cpp.txt)

Observed on NRF52_DK with/without RTOS and other (i.e network stack).
Its enough to import blinky example and replace main.cpp with provided source code.
Error is observed only with active Timeout object and maybe it is some kind of race condition.

    [ ] Question
    [ ] Enhancement
    [x ] Bug

hesolium, 2019-04-02:
I have my first thoughts on this issue. Maybe this is related to the calibration of the clocks. Both systems (STM, NRF) have special clock calibration procedures. Maybe we should ask a hardware consultant.
If this behavior is related to calibration (not ordinary bug:), I see one solution. In the 'read_high_resolution_us' procedure, we must remember the last read value of the clock and compare it with the current one at the next reading. If the previous value is larger, then it must be omitted and the procedure should return (as the current reading) the previous value + additional quantum of time. By this method, we will achieve the desired effect (always run time forward).

lrusinowicz, 2019-04-03:
This is just a bug in your program.
I'm not sure what you are trying to achieve, but you are calling `readTimer()`, which operates on global, non-volatile variables both, from interrupt and non-interrupt context, without using any kind of interlocking. This is definitely unsafe and results exactly in the effect you are describing.
When the timer interrupt hits when you are in the middle of `readTimer()` (non-interrupt context), after reading the timer but before comparison, the function has old value of timer already stored in registers. But the interrupt updates `lastReaded` to yet newer value, and the comparison will return `true`.
For this to work you'd have to put timer reading and comparison within critical section.

hesolium, 2019-04-03:
You're right, I'm sorry. When I remove function 'readTimer' from the program and call 'read_high_resolution_us' directly, everything is OK. This is typical 'race condition' (in my program:).

closed 2019-04-03 by hesolium