ARMmbed/mbed-os #10300: Support external storage type out of mbed-os tree in KVStore

ccli8, 2019-04-03:
### Description

**mbed-os**
5.12.0

Checking KVStore released with Mbed OS 5.12, I find that for `storage_type` like `TDB_EXTERNAL`, `TDB_EXTERNAL_NO_RBP`, `FILESYSTEM`, and `FILESYSTEM_NO_RBP`, only mbed-os in-tree block devices are supported:

**mbed-os/features/storage/kvstore/conf/filesystem/mbed_lib.json**:
<pre>
{
    "name": "storage_filesystem",
    "config": {
        "rbp_internal_size": {
            "help": "Default is the size of the 2 last sectors of internal flash",
            "value": "0"
        },
        "internal_base_address": {
            "help": "If default, base address is the first sector after the application code",
            "value": "0"
        },
        "filesystem": {
            "help": "Options are default, FAT or LITTLE. If default value the filesystem is chosen by the blockdevice type",
            "value": "default"
        },
        <b>
        "blockdevice": {
            "help": "Options are default, SPIF, DATAFASH, QSPIF or SD. If default, the block device will be chosen according to the component defined in targets.json",
            "value": "default"
        },
        </b>
</pre>

Would you plan to support block device out of mbed-os tree like Nuvoton's [NuMaker-mbed-SD-driver](https://github.com/OpenNuvoton/NuMaker-mbed-SD-driver) block device?

### Issue request type

    [ ] Question
    [X] Enhancement
    [ ] Bug

0xc0170, 2019-04-03:
cc @ARMmbed/mbed-os-storage

davidsaada, 2019-04-03:
@ccli8 - we have no such plans. This would be quite hard to add mainly due to testing reasons. One would need to override the `kv_init_storage_config` function (it's a weak function) and basically mimic what it does with your own block device (code can be seen in [kv_config.cpp](https://github.com/ARMmbed/mbed-os/blob/master/features/storage/kvstore/conf/kv_config.cpp)).

closed 2019-04-18 by ciarmcom

linlingao, 2019-04-30:
@ccli8 Please reopen if you have further questions.

ccli8, 2019-05-02:
#10355 is raised to support this feature. So #10300 here can be closed.