ARMmbed/mbed-os #10305: LoRaWAN public/private network selection

Luka234, 2019-04-03:
### Description

Public/private network is configured with 'public-network' parameter in mbed_lib.json of lorawan library. I don't see option to change network type with API. What is intended/recommended way of implementing nodes with run-time configurable network type?

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

loverdeg-ep, 2019-04-04:
Have you come across any further documentation for how private networks work with the LoRa radios?

I'm not sure how the radio handles receiving frames once it detects a frame that doesn't match the network "code" it is set for.

Luka234, 2019-04-05:
As I understand, radio will be blind (preamble mismatch) for not selected
network.
I have trouble understanding intended use-case and API of lorawan library
(why that configuration and couple others can't be changed at runtime).

Luka234, 2019-04-05:
I should say radio sync-word mismatch

closed 2019-04-18 by ciarmcom

linlingao, 2019-04-30:
Please reopen if you're expecting more response on this issue.