ARMmbed/mbed-os #10306: LoRaWAN: Uplink counter increments twice when Confirmed Downlink is received

reissjason, 2019-04-03:
### LoRaWAN stack - Uplink counter increments twice when Confirmed Downlink is received

LoRaMac::post_process_mcps_req is fired multiple times incrementing the counter each time.
At the network server it looks as if a packet was missed.

In the following example FCNT 50 was never transmitted.

>  Received message from Network Server 
>  RX Data on port 1 (3 bytes): 12 31 23 
> post_process_mcps_req
> CNF increment counter 49
> post_process_mcps_req
> CNF increment counter 50
> 
>  Message Sent to Network Server 
> 
>  Dummy Sensor Value = 46.1 
> 
>  26 bytes scheduled for transmission 
> post_process_mcps_req
> UNCNF increment counter 51

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ x] Bug

hasnainvirk, 2019-04-10:
@reissjason Which branch are you testing with ? Master ?

reissjason, 2019-04-10:
The project was built with mbed-os-5.11.5

https://github.com/MultiTechSystems/mbed-os-lorawan-tinyshell

reissjason, 2019-04-10:
Device was using Class C mode. Class A did not show this behavior.
Tested with master with same results. Appears to happen only when received in Class C window.

AnttiKauppila, 2019-09-13:
“At this time, due to our prioritisation of work on customer projects, we are unable to devote the time to fix this issue. As an open source project, we welcome fixes to Mbed OS, so if would like to contribute a fix yourself then we would very much welcome that, please see our contribution guidelines here - https://os.mbed.com/docs/mbed-os/v5.14/contributing/workflow.html#contributions”

closed 2019-09-13 by ciarmcom