ARMmbed/mbed-os #10307: LoRaWAN: MAC responds to Confirmed Downlink with ACK in Confirmed Uplink

reissjason, 2019-04-03:
### Description

When the end-device receives a Confirmed Downlink the MAC will send the ACK in a Confirmed Uplink which then causes a downlink from the network server to return an ACK. 

There is no reason to request an ACK for an ACK.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ x] Bug

reissjason, 2019-04-10:
Device was using Class C mode. Class A did not show this behavior.

reissjason, 2019-04-10:
Tested on mbed-os-5.11.5 and master (ffe9ddfb2ded8e601a62d3fb1ad9adadd97f4bda)

AnttiKauppila, 2019-09-13:
“At this time, due to our prioritisation of work on customer projects, we are unable to devote the time to fix this issue. As an open source project, we welcome fixes to Mbed OS, so if would like to contribute a fix yourself then we would very much welcome that, please see our contribution guidelines here - https://os.mbed.com/docs/mbed-os/v5.14/contributing/workflow.html#contributions”

closed 2019-09-13 by ciarmcom