ARMmbed/mbed-os #10312: UBLOX_EVK_ODIN_W2 with IAR returns DEVICE_ERROR instead of AUTH_ERROR/NO_CONNECTION when wrong channel is set

michalpasztamobica, 2019-04-04:
### Description

Target: UBLOX_EVK_ODIN_W2
Toolchain: IAR 8
Tools: mbed-cli
SHA:
Steps to reproduce:
```
mbed test --compile -t IAR -m UBLOX_EVK_ODIN_W2 -n tests-network-wifi --app-config=mbed_app.wlan0.json -v
mbed test --run -t IAR -m UBLOX_EVK_ODIN_W2 -n tests-network-wifi -v
```
Expected result: All tests passing
Actual result:
```
        test case: 'WIFI-CONNECT-PARAMS-CHANNEL-FAIL' ................................................ FAIL in 20.74 sec
        test case: 'WIFI-CONNECT-SECURE-FAIL' ........................................................ FAIL in 21.46 sec

```
Further details:
The issue is visible in [mbed-os nightly test runs](https://mbed-os.mbedcloudtesting.com/view/mbed-os-nightly/job/mbed-os-ci_greentea-test-IPCore-nightly/16/testReport/junit/UBLOX_EVK_ODIN_W2-IAR/tests-network-wifi/UBLOX_EVK_ODIN_W2_IAR8___WIFI_CONNECT_PARAMS_CHANNEL_FAIL/) and also reproducible on custom Wi-Fis.

These are the exact checks that fail: [wifi_connect_secure_fail](https://github.com/ARMmbed/mbed-os/blob/d1da622797773ecde10e96a65ef08f87ce6e8923/TESTS/network/wifi/wifi_connect_secure_fail.cpp#L36) and [wifi_connect_params_channel_fail](https://github.com/ARMmbed/mbed-os/blob/d1da622797773ecde10e96a65ef08f87ce6e8923/TESTS/network/wifi/wifi_connect_params_channel_fail.cpp#L39).

I have checked mbed-os code down until [this line](https://github.com/ARMmbed/mbed-os/blob/d1da622797773ecde10e96a65ef08f87ce6e8923/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F439xI/TARGET_MODULE_UBLOX_ODIN_W2/sdk/ublox-odin-w2-drivers/OdinWiFiInterface.cpp#L1274):
```
void OdinWiFiInterface::handle_wlan_status_disconnected(void)
{
...
			switch(_wlan_status_disconnected_info) {
...
				default:
					error_code = NSAPI_ERROR_DEVICE_ERROR;
					break;
			}
			send_user_response_msg(ODIN_WIFI_MSG_USER_CONNECT, error_code);
```
To my understanding the error code comes from the library that we have no access to.
It is important to note that neither GCC nor ARM don't show this behaviour and that the [static library for IAR](https://github.com/ARMmbed/mbed-os/tree/master/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F439xI/TARGET_MODULE_UBLOX_ODIN_W2/sdk) was updated a month ago.
@ARMmbed/team-ublox , could you please check this issue?
@SeppoTakalo , @TuomoHautamaki , FYI.
### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

AmmadRehmat, 2019-04-11:
Thanks for pointing out, we have fixed the issue and will be included in our next release. 
https://jira.u-blox.net/browse/TE_OM2-278

0xc0170, 2019-06-07:
Should be fixed by 10312. If not, reopen with an update

closed 2019-06-07 by 0xc0170

michalpasztamobica, 2019-06-07:
Let me explicitly confirm. Our CI shows a pass for UBLOX Wi-Fi now with [this revision of mbed-os](https://github.com/ARMmbed/mbed-os/#73f096399b4cda1f780b140c87afad9446047432) and I checked that there were some updates from Ublox team since March, which could be making the difference.
@ARMmbed/team-ublox , thanks for fixing!