ARMmbed/mbed-os #10313: Build issue with STM32F746

jeromecoutant, 2019-04-04:
### Description

I am checking NUCLEO_F746ZG build, but I think issue is larger than this...

- question on MBED_BOOT_STACK_SIZE:

https://github.com/ARMmbed/mbed-os/blob/d1da622797773ecde10e96a65ef08f87ce6e8923/targets/TARGET_STM/TARGET_STM32F7/TARGET_STM32F746xG/device/TOOLCHAIN_ARM_STD/stm32f746xg.sct#L40

These lines ar not used at all, as MBED_BOOT_STACK_SIZE default value is set in the targets.json:
````
            "boot-stack-size": {
                "value": "0x1000"
````

- question on MBED_RAM_START and MBED_RAM_SIZE:

https://github.com/ARMmbed/mbed-os/blob/d1da622797773ecde10e96a65ef08f87ce6e8923/targets/TARGET_STM/TARGET_STM32F7/TARGET_STM32F746xG/device/TOOLCHAIN_ARM_STD/stm32f746xg.sct#L45

Seems that these values are predefined thanks to pack manager ?
But as STM32F746ZGTx has 2 IRAM, only 1 is taken into account, and values becomes wrong...

@theotherjimmy 
@mprse 
@VVESTM 
@LMESTM 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-04-04:
Looking at https://os.mbed.com/docs/mbed-os/latest/porting/porting-bootstrap.html, mbed boot stack size is defined by target (as in docs, by tools), the scatter file defines it only to check if it is defined as protection.

RAM start and size should be defined by tools. 

@SenRamakri  Please review as well

mprse, 2019-04-04:
> These lines ar not used at all, as MBED_BOOT_STACK_SIZE default value is set in the targets.json

This is correct, but in case the stack size is defined here (in case of build with RTOS):
https://github.com/ARMmbed/mbed-os/blob/2aa559c7a1beace717cc1361eda3a0d89eaa210c/rtos/mbed_lib.json#L28-L30

jeromecoutant, 2019-04-04:
MBED_BOOT_STACK_SIZE is now followed by #10319 

About MBED_RAM_START, should I suppose that tools/arm_pack_manager/index.json file
will not change about IRAM1/IRAM2 choice ?

Then I need to update the scatter files where MBED_RAM_START is not the correct value to choose MBED_RAM1_START value?

theotherjimmy, 2019-04-04:
@jeromecoutant That information came from ST's CMSIS Pack file. Unless ST has plans to change the hardware description (which I don't advise), please update the linker script. Further `tools/arm_pack_manager/index.json` does not choose. It's `tools/config/__init__.py` that's ordering the memory.

jeromecoutant, 2019-04-05:
Hi

Could you review #10323 ?

Thx

closed 2019-06-17 by jeromecoutant