ARMmbed/mbed-os #10319: Question on MBED_BOOT_STACK_SIZE

jeromecoutant, 2019-04-04:
### Description

Question started with https://github.com/ARMmbed/mbed-os/issues/10313#issuecomment-479837196

Seen in https://os.mbed.com/docs/mbed-os/latest/porting/porting-bootstrap.html

````
#if !defined(MBED_BOOT_STACK_SIZE)
/* This value is normally defined by the tools to 0x1000 for bare metal and 0x400 for RTOS */
  #define MBED_BOOT_STACK_SIZE  0x400
#endif
````

Even if RTOS is enabled, MBED_BOOT_STACK_SIZE will not be updated.
MBED_CONF_TARGET_BOOT_STACK_SIZE is created

See in mbed_config.h:
#define MBED_CONF_TARGET_BOOT_STACK_SIZE                                      0x400                                                                                            // set by library:rtos[*]

@0xc0170 
@SenRamakri 

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

jeromecoutant, 2019-04-04:
@mprse

jeromecoutant, 2019-04-23:
Please reopen this issue :-(

loverdeg-ep, 2019-04-23:
@0xc0170

jeromecoutant, 2019-04-25:
@linlingao

linlingao, 2019-04-25:
Isn't this a bug?

0xc0170, 2019-08-13:
@jeromecoutant need more details here. Is there an issue? As referenced, my earlier comment explains how it is. What did I miss?

hugueskamba, 2020-08-17:
@evedon
Can someone please have a look at this? It is also blocking the work on CMAke that we are doing.
Where is `MBED_BOOT_STACK_SIZE` defined?

mprse, 2020-08-17:
Please take a look at this PR:
https://github.com/ARMmbed/mbed-os/pull/8039

0xc0170, 2020-08-18:
Thanks @mprse . You might know. I fount the comment with the exact question as we have : https://github.com/ARMmbed/mbed-os/pull/8039/files#r216012402 in the PR 8039. We have 2 different stack size config names . From the PR, it looks like it was intentionally done. To have `MBED_BOOT_STACK_SIZE` for mbed 2/5 configuration as they use different sizes (and even the stack has different function). 
What is `MBED_CONF_TARGET_BOOT_STACK_SIZE` then if its not used anywhere.

PR #13452 (2020-09-10): Use boot stack size from config system

### Summary of changes 

To allow overriding of the boot stack size from the Mbed configuration
system, consistently use MBED_CONF_TARGET_BOOT_STACK_SIZE rather than
MBED_BOOT_STACK_SIZE.

Fixes #10319

#### Impact of changes 

#### Migration actions required 

### Documentation 

Porting guide needs updating. PR available at https://github.com/ARMmbed/mbed-os-5-docs/pull/1362

----------------------------------------------------------------------------------------------------------------
### Pull
commit 39e69d32 (2020-09-10): Use boot stack size from config system

To allow overriding of the boot stack size from the Mbed configuration
system, consistently use MBED_CONF_TARGET_BOOT_STACK_SIZE rather than
MBED_BOOT_STACK_SIZE.

Fixes #10319
closed 2020-09-10 by 0xc0170