ARMmbed/mbed-os #10321: nRF52840 Cordio GPIO usage

marcemmers, 2019-04-05:
### Description

We have created a custom board based on the nRF52840 and are using the Cordio BLE stack as the radio driver. It seems however that the Cordio stack is controlling some of the GPIO which is interfering with our application since we have all GPIO in use.
At least P0.13 and P0.15 are affected but we are experiencing other problems as well when we enable the stack.

Why is the radio driver controlling pins in the first place?

We are using mbed-os release 5.11.5. We have exported to a custom target and are using IAR 8.30.

I've noticed that there is a pull request open for a Cordio stack update: https://github.com/ARMmbed/mbed-os/pull/10272. Are there any changes to the GPIO usage in the update?

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

donatieng, 2019-04-17:
Hi @marcemmers and thanks for raising the issue - we'll make sure this is addressed in the upcoming PR

paul-szczepanek-arm, 2019-04-17:
this is fixed in the 19.2 cordio LL, PR pending preceding PR

yogeshk19, 2019-04-19:
@donatieng , @paul-szczepanek-arm  is this slated to be fixed in the 5.12.2 release?

shane-taurean, 2019-04-19:
I'm trying the mbed-os-example-ble-LED and mbed-os-example-ble-BatteryLevel cordio example codes on the nRF52_DK (nRF52832). I get unwanted pulses on the analog pins, around 3 pulses repeated every second during advertising and 1 pulse every 45 ms after connecting via nRFConnect.

I tried mbed-os releases 5.11.5, 5.12.1 and also the master branch.

Is this related to the same issue and any resolution expected soon?

donatieng, 2019-04-25:
@shane-taurean that looks like the same issue

Work is being done here to resolve it: https://github.com/ARMmbed/mbed-os/pull/10272

closed 2019-05-23 by ciarmcom

yogeshk19, 2019-06-13:
@donatieng When is this fix due to be available in mbed-os. I thought a new version of the mbed-os is released every other Wednesday and I was expecting a release today, since the last one was released around 5/22?

paul-szczepanek-arm, 2019-06-13:
this will be solved in 5.13 keep an eye on the main page for release announcement

yogeshk19, 2019-06-13:
@paul-szczepanek-arm thanks for taking the time to respond. Can you provide an approximate timeline when 5.13 will be released? This one issue is currently blocking us from going to pre-production.

adbridge, 2019-06-13:
5.13 is currently in the latter stages of release testing.

yogeshk19, 2019-06-13:
@adbridge is this the next slated release or is there a release in between, 5.12.5? I am just trying to figure how much of delay would it be for us, before we are able to consume this fix?

Thanks,
Yogesh

0xc0170, 2019-06-14:
@yogeshk19  rc3 still in the air, the next week , only 5.13.x (no release for 5.12 release branch)