ARMmbed/mbed-os #10327: Can we disable STDIO printf but keep DEVICE_SERIAL

LMESTM, 2019-04-05:
### Description

In https://github.com/ARMmbed/mbed-os/blob/master/platform/mbed_retarget.cpp#L277, as soon as DEVICE_SERIAL is supported, an STDIO serial will be created for printf support.

Could there be an option for disabling STDIO console and select SINK one instead, without disabling DEVICE_SERIAL which may be needed by peripherals drivers.

Could be something like:
```#if (DEVICE_SERIAL &&  !MBED_CONF_PLATFORM_STDIO_SINK))```

This could be useful when switching to release mode or to improve low power consumption.
Maybe there is already an alternative method that I am not aware of. 

### Issue request type
    [x ] Question
    [ ] Enhancement
    [ ] Bug

jeromecoutant, 2019-04-05:
In #5658 old issue, I thought it was DEVICE_STDIO_MESSAGES goal.
But I didn't get any clear answer...

0xc0170, 2019-04-05:
cc @kjbracey-arm

kjbracey-arm, 2019-04-05:
I don't understand the intent of `DEVICE_STDIO_MESSAGES` either. Presumably it was supposed to be something like this, but I'd be inclined not to touch it, as I don't know who's using it.

If for power, as of 5.12, you can save power by suspending input, although that doesn't help your ROM space: #9797

I'd suggest a new option `platform.stdio-serial` which defaults to `true`, so it's `#if  MBED_CONF_PLATFORM_STDIO_SERIAL && DEVICE_SERIAL`.

I'm reluctant to have a bunch of options for possible modes, as we already have a functional target override system I want people to use, but having an option to nix the default use-serial behaviour makes sense.

LMESTM, 2019-04-05:
What I'm actually looking for is stdio_uart to not be initialized at all.

> I'd suggest a new option `platform.stdio-serial` which defaults to `true`, so it's `#if MBED_CONF_PLATFORM_STDIO_SERIAL && DEVICE_SERIAL`.

So yes, that sounds good to me .

PR #10328 (2019-04-09): Add option to disable default UART console
### Description

New `target.console-uart` option added to indicate whether a target has a console UART on `STDIO_UART_TX`/`RX`/`RTS`/`CTS` pins. (The existing option `target.console-uart-flow-control` indicates whether RTS and or CTS is available in addition to TX and RX).

The option defaults to true, and is currently true on all platforms. It only applies if `DEVICE_SERIAL` is true, so no need to go through and mark it false for non-SERIAL platforms.

An application can turn off `target.console-uart` to save ROM/power/etc if they don't want to use  [... cut]
closed 2019-04-09 by cmonr