ARMmbed/mbed-os #10331: update icf files for all targets with proper __CRASH_DATA_ symbols

studavekar, 2019-04-05:
### Description

The fix which was introduced in https://github.com/ARMmbed/mbed-os/pull/10311 , need to be done across all targets supporting the crash report feature.

Note : Failure are not since for other platform however it would be worth to update icf files appropriately.

### Issue request type

    [ ] Question
    [x] Enhancement
    [ ] Bug

studavekar, 2019-04-05:
Here is the list of devices 

```json
        "DISCO_L475VG_IOT01A": {
            "crash-capture-enabled": true,
            "fatal-error-auto-reboot-enabled": true
        },
        "K64F": {
            "crash-capture-enabled": true,
            "fatal-error-auto-reboot-enabled": true
        },
        "K66F": {
            "crash-capture-enabled": true,
            "fatal-error-auto-reboot-enabled": true
        },
        "NUCLEO_F429ZI": {
            "crash-capture-enabled": true,
            "fatal-error-auto-reboot-enabled": true
        },
        "NUCLEO_F746ZG": {
            "crash-capture-enabled": true,
            "fatal-error-auto-reboot-enabled": true
        },
        "NUCLEO_F767ZI": {
            "crash-capture-enabled": true,
            "fatal-error-auto-reboot-enabled": true
        },
        "NUCLEO_H743ZI": {
            "crash-capture-enabled": true,
            "fatal-error-auto-reboot-enabled": true
        },
        "NUCLEO_F439ZI": {
            "crash-capture-enabled": true,
            "fatal-error-auto-reboot-enabled": true
        },
        "UBLOX_EVK_ODIN_W2": {
            "crash-capture-enabled": true,
            "fatal-error-auto-reboot-enabled": true
        },
        "UBLOX_C030_U201": {
            "crash-capture-enabled": true,
            "fatal-error-auto-reboot-enabled": true
        },
        "NUMAKER_PFM_M487": {
            "crash-capture-enabled": true,
            "fatal-error-auto-reboot-enabled": true
        },
        "NRF52840_DK": {
            "crash-capture-enabled": true,
            "fatal-error-auto-reboot-enabled": true
        },
        "NUCLEO_L476RG": {
            "crash-capture-enabled": true,
            "fatal-error-auto-reboot-enabled": true
        },
        "NUCLEO_F411RE": {
            "crash-capture-enabled": true,
            "fatal-error-auto-reboot-enabled": true
        }
```

0xc0170, 2019-04-05:
cc @SenRamakri

jeromecoutant, 2019-04-05:
Hi

Ideal solution is to make all icf ready for crash feature...

User just has to enable "crash-capture-enabled" in their mbed_app.json,
and it works! (without updating its own icf)

jeromecoutant, 2019-04-23:
Please reopen this issue!

jeromecoutant, 2019-04-25:
@linlingao

closed 2019-10-31 by ciarmcom