ARMmbed/mbed-os #10336: [Error] mbed_rtc_time.cpp@89,22: invalid use of incomplete type 'const struct timeval'

Hoel, 2019-04-07:
### Description

When i create a new project from CLI 

`mbed new new_project`

and try to compile 

`mbed compile -m XDOT_L151CC -t GCC_ARM`

i got this error:

`Compile [ 81.7%]: mbed_rtc_time.cpp
[Error] mbed_rtc_time.cpp@89,22: invalid use of incomplete type 'const struct timeval'
[Error] mbed_rtc_time.h@111,25: forward declaration of 'const struct timeval'
[Error] mbed_rtc_time.cpp@110,7: invalid use of incomplete type 'struct timeval'
[Error] mbed_rtc_time.h@111,25: forward declaration of 'struct timeval'
[Error] mbed_rtc_time.cpp@111,7: invalid use of incomplete type 'struct timeval'
[Error] mbed_rtc_time.h@111,25: forward declaration of 'struct timeval'
[Error] mbed_rtc_time.cpp@124,20: aggregate 'timeval tv' has incomplete type and cannot be defined
[Error] mbed_rtc_time.cpp@137,26: variable 'const timeval tv' has initializer but incomplete type
[Warning] mbed_rtc_time.cpp@132,1: control reaches end of non-void function [-Wreturn-type]
[ERROR] ../mbed-os/platform/mbed_rtc_time.cpp: In function 'int settimeofday(const timeval*, const timezone*)':
../mbed-os/platform/mbed_rtc_time.cpp:89:22: error: invalid use of incomplete type 'const struct timeval'
         _rtc_write(tv->tv_sec);
                      ^
In file included from ../mbed-os/platform/mbed_rtc_time.cpp:20:0:
../mbed-os/platform/mbed_rtc_time.h:111:25: error: forward declaration of 'const struct timeval'
 int gettimeofday(struct timeval *tv, void *tz);
                         ^
../mbed-os/platform/mbed_rtc_time.cpp: In function 'int gettimeofday(timeval*, void*)':
../mbed-os/platform/mbed_rtc_time.cpp:110:7: error: invalid use of incomplete type 'struct timeval'
     tv->tv_sec  = t;
       ^
In file included from ../mbed-os/platform/mbed_rtc_time.cpp:20:0:
../mbed-os/platform/mbed_rtc_time.h:111:25: error: forward declaration of 'struct timeval'
 int gettimeofday(struct timeval *tv, void *tz);
                         ^
../mbed-os/platform/mbed_rtc_time.cpp:111:7: error: invalid use of incomplete type 'struct timeval'
     tv->tv_usec = 0;
       ^
In file included from ../mbed-os/platform/mbed_rtc_time.cpp:20:0:
../mbed-os/platform/mbed_rtc_time.h:111:25: error: forward declaration of 'struct timeval'
 int gettimeofday(struct timeval *tv, void *tz);
                         ^
../mbed-os/platform/mbed_rtc_time.cpp: In function 'time_t time(time_t*)':
../mbed-os/platform/mbed_rtc_time.cpp:124:20: error: aggregate 'timeval tv' has incomplete type and cannot be defined
     struct timeval tv;
                    ^
../mbed-os/platform/mbed_rtc_time.cpp: In function 'void set_time(time_t)':
../mbed-os/platform/mbed_rtc_time.cpp:137:26: error: variable 'const timeval tv' has initializer but incomplete type
     const struct timeval tv = { t, 0 };
                          ^
../mbed-os/platform/mbed_rtc_time.cpp: In function 'time_t time(time_t*)':
../mbed-os/platform/mbed_rtc_time.cpp:132:1: warning: control reaches end of non-void function [-Wreturn-type]
 }`

 So it complains about const struct timeval in mbed_rtc_time.cpp, the target is XDOT_L151CC.
Nothing else was added in the project other than a main.cpp blinking a LED so there is something wrong with mbed-os itself.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

Hoel, 2019-04-08:
follow-up on the issue, the problem has been solved after i changed the toolchain, i installed mbed-cli-osx-installer from theotherjimmy  https://github.com/ARMmbed/mbed-cli-osx-installer/releases/tag/v0.0.10 and the error went away

closed 2019-04-08 by Hoel

ccu1tn, 2019-05-13:
Dear @Hoel 
I have the same your errors. 
[image]
I was download mbed-cli-osx-installer file. 
I trying to install it on Ubuntu, but it not work.
so, can you tutorial me how to install and config it?
many thank

Hoel, 2019-05-15:
@ccu1tn 
Hello, well, you have the answer right before the question, this problem was toolchain related, so if you have this problem (with another os) then, its most likely also toolchain related. So you should check which GCC you have, check if the installer from TheOtherJimmy is available for Linux, otherwsie check inside which GCC is installed and install it on your Linux.