ARMmbed/mbed-os #10337: STM32 stuck in sleep

masoudr, 2019-04-08:
### Description

Hi,
I'm trying to test the functionality of `sleep` function on my STM32F407 target with `mbed-os 5.12.0`. I would want to force the device to go into deep sleep mode with a specific period of time.
I've also enabled `Tickless` in the `target.json` and removed `platform.stdio-buffered-serial` from device configuration as stated in the [docs](https://os.mbed.com/docs/mbed-os/v5.12/apis/power-management.html#sleep).
My problem is when I use [this ](https://os.mbed.com/users/Sissors/code/WakeUp/)Wakeup library my device never wakes up from sleep.
I'm not sure if I miss something in the configuration also the [LowPowerTimeout API](https://os.mbed.com/docs/mbed-os/v5.9/reference/lowpowertimeout.html) didn't work for me (Call back function never executed in time intervals).
```
#include "mbed.h"
#include "WakeUp.h"
 
DigitalOut myled(LED1);
 
int main() {
    //The low-power oscillator can be quite inaccurate on some targets
    //this function calibrates it against the main clock
    WakeUp::calibrate();
   
    while(1) {
        //Set LED to zero
        myled = 0;
        
        //Set wakeup time for 2 seconds
        WakeUp::set_ms(2000);
        
        //Enter deepsleep, the program won't go beyond this point until it is woken up
        sleep();
  
        //Set LED for 1 second to one
        myled = 1;
        wait(1);
    }
}
```
### Issue request type

    [x] Question
    [ ] Enhancement
    [x] Bug

LMESTM, 2019-04-08:
@masoudr I don't know the wakeup library - you'd have to ask the author, I think it's @Sissors  ?
If you want to test sleep or deep sleep, I'd recommend to simply use 
https://github.com/ARMmbed/mbed-os-example-blinky
This basic example will report the time spent in sleep mode. Can you try it out and see if the results match you needs ? 
Note that mbed-os will wake-up regularly from sleep or deep-sleep to maintain the list of running timers, so you will find regular small wake-up times.

Sissors, 2019-04-08:
That is indeed my library. But it has been a long time ago I have looked into this, and I am not really working on it (or mbed in general) anymore. Especially with all the changes to sleep timers that have been done in mbed (-os) it is likely stuff has been broken*. In general it has been intended for 'plain' mbed and not mbed-os, since that one has as @LMESTM mentions a lot of stuff happening on the background. 

* In general most of my libraries I have built on top of mbed libraries, and the WakeUp lib for STM is included in this, it re-uses things from the RTC library. The advantage of this is less work for me, and way more devices supported than if I had to do everything manually. The disadvantage is that mbed updates break stuff.

masoudr, 2019-04-08:
@LMESTM @Sissors Thanks, but as I already said I want to force my device into deep sleep mode for power saving and then it should automatically wake up after timeout reached (not on interrupts or external signals). I only find the `Wakeup` library to do such a thing. If you know any other working way I would appreciate.
I think putting a simple `wait` function is not supposed to put the device in deep sleep.

LMESTM, 2019-04-08:
@masoudr in mbed-os, it is the sleep managers that determines when system can go to deep sleep.
see https://www.youtube.com/watch?v=OFfOlBaegdg

I think that what you're looking for is an RTC alarm API, which doesn't really exist as such in mbed (@ARMmbed/mbed-os-core 1 more request for RTC alarm API). 

But you can find an alarm example program here: 
https://github.com/ARMmbed/mbed-os-example-alarm/blob/master/main.cpp

linlingao, 2019-04-09:
Unchecked Bug. Appears to be a question.

masoudr, 2019-04-12:
@LMESTM Thanks, I figured it out, as in the video stated, it seems that just terminating all threads and using sleep will do the job, I wrote a function like below, which seems put the device into sleep, but got 2 more questions:
1. Do I really need to enable `Tickless`?
2. It seems that putting the device into a deep sleep is handle by `mbed-os` itself so which params prevent device to get into deep sleep?
I tried to terminate all my threads but still seems that missing something else to prepare for a deep sleep.

```
void sleep_helper(int seconds)
{
    int i = 0;
    const int timeout = seconds * 50;
    while (i < timeout)
    {
        wait_ms(20);
        i++;
    }
}
```

closed 2019-04-18 by ciarmcom

jeromecoutant, 2019-04-23:
@masoudr  do you want to reopen this issue ?

LMESTM, 2019-04-23:
> * Do I really need to enable `Tickless`? What is the point of enabling it?
Without tickless, there will be TICKs every 1 ms. So you would not have much time to sleep between each tick

> * It seems that putting the device into a deep sleep is handle by `mbed-os` itself so which params prevent device to get into deep sleep?

As long as there is some activity (UART, High resolution timers, active threads, etc ... ) there will not be any deep sleep. There is also an API (lock_deep_sleep / unlock_deep_sleep) that allows you to prevent deep sleep entry when you don't want it.

masoudr, 2019-04-23:
@jeromecoutant No doesn't need to, I somehow put the device into sleep mode.
@LMESTM I just terminated all threads (except the LowPowerTicker), turned off UART devices and used the above function, thus I saw that the device become in a low power mode but I'm not sure if it is in sleep or deep sleep.
Now I'm trying to use `unlock_deep_sleep` to somehow save more energy by putting the device into deep sleep mode.

LMESTM, 2019-04-23:
@masoudr You don't need to call unlock_deep_sleep unless you have called lock_deep_sleep in a first place... mbed-os offers statistics if the time spent in sleep and deep sleep. you can look for documentation of mbed_stats_cpu_get(&cpu_stats); API.

linlingao, 2019-04-30:
@masoudr Please reopen should you have further questions.