ARMmbed/mbed-os #10345: ST Nucleo F401RE Deep sleep

ventus2c, 2019-04-08:
### Description

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-04-08:
Hi, please fill in the details (description, Issue type), otherwise this will be closed

ventus2c, 2019-04-08:
Hello,
I’m using nucleo blink led example with mbed-os 5-12 to test this board consumption. I connected a multimeter across jumper IDD.
I’m using online compiler and I compiled using this macros:
DEVICE_SLEEP
MBED_TICKLESS
DEVICE_LPTICKER
MBED_SLEEP_TRACING_ENABLED

In mbed_app.json in device_override:
            "platform.stdio-buffered-serial":false 

Here is the tracing result:
*Sleep locks held:
*[id: mbed_rtx_idle., count: 1]
*UNLOCK: mbed_rtx_idle.cpp, ln: 133, lock count: 0
*LOCK: mbed_rtx_idle.cpp, ln: 111, lock count: 1
With LED off I measured a current of 5.76mA, that is far too much considering that it should go deepsleep.

I tried also adding:
MBED_CONF_TARGET_TICKLESS_FROM_US_TICKER=0
Nothing happens, it seems it's not going to deepsleep
What could be wrong?
Mario

closed 2019-04-18 by ciarmcom

jeromecoutant, 2019-04-23:
@ventus2c do you want to reopen this issue ?

ventus2c, 2019-04-23:
I used MBED_CONF_TARGET_TICKLESS_FROM_US_TICKER=0 in MACROS section of online compiler, didn't work.
I tryed in mbed_app.json macros "MBED_CONF_TARGET_TICKLESS_FROM_US_TICKER=0", WORKED.
Consider it closed.