ARMmbed/mbed-os #10351: 'mbed import' could not import from https://os.mbed.com

turmary, 2019-04-09:
### Description

Mbed CLI on Windows

version: https://github.com/ARMmbed/mbed-cli-windows-installer/releases/download/v0.4.10/Mbed_installer_v0.4.10.exe

tary@tary-think MINGW32 /f/home/tary/mbed
$ git clone https://os.mbed.com/users/hudakz/code/CAN_Hello
Cloning into 'CAN_Hello'...
fatal: unable to update url base from redirection:
  asked for: https://os.mbed.com/users/hudakz/code/CAN_Hello/info/refs?service=git-upload-pack
   redirect: https://os.mbed.com/users/hudakz/code/CAN_Hello/info/refs/?service=git-upload-pack

tary@tary-think MINGW32 /f/home/tary/mbed
$ git clone http://os.mbed.com/users/hudakz/code/CAN_Hello
Cloning into 'CAN_Hello'...
fatal: unable to update url base from redirection:
  asked for: http://os.mbed.com/users/hudakz/code/CAN_Hello/info/refs?service=git-upload-pack
   redirect: https://os.mbed.com/users/hudakz/code/CAN_Hello/info/refs/?service=git-upload-pack

tary@tary-think MINGW32 /f/home/tary/mbed
$ mbed import http://os.mbed.com/users/hudakz/code/CAN_Hello
[mbed] Working path "F:\home\tary\mbed" (directory)
[mbed] Program path "F:\home\tary\mbed"
[mbed] Importing program "CAN_Hello" from "https://os.mbed.com/users/hudakz/code/CAN_Hello" at latest revision in the current branch
[mbed] ERROR: Unable to clone repository (http://os.mbed.com/users/hudakz/code/CAN_Hello)

tary@tary-think MINGW32 /f/home/tary/mbed
$ mbed import https://os.mbed.com/users/hudakz/code/CAN_Hello
[mbed] Working path "F:\home\tary\mbed" (directory)
[mbed] Program path "F:\home\tary\mbed"
[mbed] Importing program "CAN_Hello" from "https://os.mbed.com/users/hudakz/code/CAN_Hello" at latest revision in the current branch
[mbed] ERROR: Unable to clone repository (https://os.mbed.com/users/hudakz/code/CAN_Hello)

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

cmonr, 2019-04-09:
@turmary Could you run `git --version` on your system to verify that it's setup correctly?

turmary, 2019-04-10:
Hi cmonr,

tary@tary-think MINGW32 /f/home/tary/mbed
$ git clone https://os.mbed.com/users/hudakz/code/CAN_Hello
Cloning into 'CAN_Hello'...
fatal: unable to update url base from redirection:
  asked for: https://os.mbed.com/users/hudakz/code/CAN_Hello/info/refs?service=git-upload-pack
   redirect: https://os.mbed.com/users/hudakz/code/CAN_Hello/info/refs/?service=git-upload-pack

tary@tary-think MINGW32 /f/home/tary/mbed
$ git --version
git version 2.17.1.windows.2

tary@tary-think MINGW32 /f/home/tary/mbed
$ which git
/mingw32/bin/git

tary@tary-think MINGW32 /f/home/tary/mbed
$ echo $PATH
/mingw32/bin:/usr/local/bin:/usr/bin:/bin

I tried decrease my PATH setting, but the error still exist.

malavikasajikumar, 2019-09-04:
Hello, has there been a solution to this yet? Looks like 'mbed import' command works only on certain https://os.mbed.com/teams code.

0xc0170, 2019-09-05:
@malavikasajikumar Can you post specific examples that fail?

closed 2020-02-18 by ciarmcom

turmary, 2020-04-12:
Finally I realized the repository located at https://os.mbed.com/users/XXX/code
are managed by Mercurial.
The import error was introduced by hg/Mercurial.
The hg installed by Mbed_installer_v0.4.10.exe lacks web.cacerts configuration, caused SSL certification failed when run `hg import`.