ARMmbed/mbed-os #10352: ATHandler.cpp malloc undefined

wajahat-ublox, 2019-04-09:
### Description
I am getting build errors if cellular traces are enabled. The error points to [this ](https://github.com/ARMmbed/mbed-os/blob/aff381579034006ece5c7df37633f2bca95f162b/features/cellular/framework/AT/ATHandler.cpp#L1292)line. Wouldn't it be better to use new/delete instead of malloc free as we are in .cpp?

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-04-09:
cc @ARMmbed/mbed-os-wan

hasnainvirk, 2019-04-09:
@wajahat-ublox Interesting. It shouldn't matter how you allocate memory. Malloc is just as fine as new although there morphology is different. Are you sure you are not running out of memory at the point you twiddle the traces on ? May I ask the name and specs of the board you are using ?

hasnainvirk, 2019-04-09:
Please share your error log as well.

wajahat-ublox, 2019-04-09:
@hasnainvirk 
Yes technically both are same, in fact i think mbed calls malloc from new. However if i replace the malloc with new calls, then it builds fine.

I don't think i am running out of memory because i am talking about compilation error while building with traces enabled.

```
Building project dl_test (UBLOX_C027, ARMC5)
Scan: dl_test
Compile [100.0%]: ATHandler.cpp
[Error] ATHandler.cpp@1292,0:  #20: identifier "malloc" is undefined
[Error] ATHandler.cpp@1322,0:  #20: identifier "free" is undefined
[ERROR] ".\mbed-os\features\cellular\framework\AT\ATHandler.cpp", line 1292: Error:  #20: identifier "malloc" is undefined
".\mbed-os\features\cellular\framework\AT\ATHandler.cpp", line 1322: Error:  #20: identifier "free" is undefined
.\mbed-os\features\cellular\framework\AT\ATHandler.cpp: 0 warnings, 2 errors
```

hasnainvirk, 2019-04-09:
@wajahat-ublox Alright. Thanks, one way to fix it would be to add stdlib.h explicitly in ATHandler.cpp. But I like your approach. Let's use new instead.

closed 2019-04-09 by wajahat-ublox