ARMmbed/mbed-os #10356: Renesas GR-PEACH (RZ_A1H) Greentea tests fail with ARMC6

juhaylinen, 2019-04-09:
### Description

**Target**: RZ_A1H
**Mbed OS**: dc1198b5c8

Greentea tests fail to sync error when ARMC6 (ARM Compiler 6.12) is used.
No issue with ARMC5, GCC_ARM or IAR8.

Log:
```
[1554779884.37][GLRM][INF] opening connection to platform at baudrate='9600'
[1554779884.61][GLRM][INF] remote resources reset...
[1554779888.00][GLRM][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
[1554779888.00][CONN][INF] sending up to 10 __sync packets (specified with --sync=10)
[1554779888.00][CONN][INF] sending preamble '0fbfe12d-7eea-43b3-883c-942c7e1cd472'
[1554779888.16][GLRM][TXD] {{__sync;0fbfe12d-7eea-43b3-883c-942c7e1cd472}}
[1554779893.16][CONN][INF] Reset the part and send in new preamble...
[1554779893.16][GLRM][INF] remote resources reset...
[1554779895.53][CONN][INF] resending new preamble '3dbaea6e-ee40-489c-8d38-7a9e96ff30d7' after 5.00 sec
[1554779896.54][GLRM][TXD] {{__sync;3dbaea6e-ee40-489c-8d38-7a9e96ff30d7}}
[1554779901.54][CONN][INF] Reset the part and send in new preamble...
[1554779901.54][GLRM][INF] remote resources reset...
[1554779903.91][CONN][INF] resending new preamble '434122e9-cae3-4dea-a96c-e98448dd159c' after 5.00 sec
[1554779904.97][GLRM][TXD] {{__sync;434122e9-cae3-4dea-a96c-e98448dd159c}}
[1554779909.98][CONN][INF] Reset the part and send in new preamble...
[1554779909.98][GLRM][INF] remote resources reset...
[1554779912.29][CONN][INF] resending new preamble '6091ef21-401b-4763-812b-1e1dbed3cdc5' after 5.00 sec
[1554779913.36][GLRM][TXD] {{__sync;6091ef21-401b-4763-812b-1e1dbed3cdc5}}
[1554779918.36][CONN][INF] Reset the part and send in new preamble...
[1554779918.36][GLRM][INF] remote resources reset...
[1554779920.67][CONN][INF] resending new preamble '90494d0e-00af-4d6e-af0f-8f06c1d68aff' after 5.00 sec
[1554779921.74][GLRM][TXD] {{__sync;90494d0e-00af-4d6e-af0f-8f06c1d68aff}}
[1554779926.74][CONN][INF] Reset the part and send in new preamble...
[1554779926.74][GLRM][INF] remote resources reset...
[1554779929.06][CONN][INF] resending new preamble 'a5f880c4-ed2f-4fb5-a33b-1f724da5b457' after 5.00 sec
[1554779930.12][GLRM][TXD] {{__sync;a5f880c4-ed2f-4fb5-a33b-1f724da5b457}}
[1554779935.12][CONN][INF] Reset the part and send in new preamble...
[1554779935.12][GLRM][INF] remote resources reset...
[1554779937.49][CONN][INF] resending new preamble '4569b708-12ba-4cef-bb48-a8106f4c3136' after 5.00 sec
[1554779938.50][GLRM][TXD] {{__sync;4569b708-12ba-4cef-bb48-a8106f4c3136}}
[1554779943.50][CONN][INF] Reset the part and send in new preamble...
[1554779943.50][GLRM][INF] remote resources reset...
[1554779945.82][CONN][INF] resending new preamble '84c448fa-5b71-4707-8b62-ea4a76338a0a' after 5.00 sec
[1554779946.89][GLRM][TXD] {{__sync;84c448fa-5b71-4707-8b62-ea4a76338a0a}}
[1554779951.89][CONN][INF] Reset the part and send in new preamble...
[1554779951.89][GLRM][INF] remote resources reset...
[1554779954.20][CONN][INF] resending new preamble '9ac1270f-34cc-45ca-963e-646210f24055' after 5.00 sec
[1554779955.27][GLRM][TXD] {{__sync;9ac1270f-34cc-45ca-963e-646210f24055}}
[1554779960.27][CONN][INF] Reset the part and send in new preamble...
[1554779960.27][GLRM][INF] remote resources reset...
[1554779962.58][CONN][INF] resending new preamble '9b7a4c40-9d38-40d5-9cda-3a6db9254f08' after 5.00 sec
[1554779963.65][GLRM][TXD] {{__sync;9b7a4c40-9d38-40d5-9cda-3a6db9254f08}}
[1554779967.07][HTST][ERR] None
[1554779967.07][HTST][WRN] stopped to consume events due to __notify_sync_failed event
[1554779967.07][HTST][INF] __exit_event_queue received
[1554779967.07][HTST][INF] test suite run finished after 110.07 sec...
[1554779967.07][HTST][INF] CONN exited with code: 0
[1554779967.07][HTST][INF] No events in queue
[1554779967.07][HTST][INF] host test result() call skipped, received: sync_failed
[1554779967.07][HTST][WRN] missing __exit event from DUT
[1554779967.07][HTST][INF] calling blocking teardown()
[1554779967.07][HTST][INF] teardown() finished
[1554779967.07][HTST][INF] {{result;sync_failed}}
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

juhaylinen, 2019-04-10:
ARMC6 debug profile seem to work other profiles not. You can produce the issue also with https://github.com/ARMmbed/mbed-os-example-blinky

0xc0170, 2019-04-10:
cc @ARMmbed/team-renesas

michalpasztamobica, 2019-04-11:
GR-LYCHEE is affected the same way.

d-kato, 2019-05-15:
This issue of GR_LYCHEE & RZ_A1H can be resolved by modifying "__FPU_Enable()" and adding FPU settings in arm.py.
Fix of "__FPU_Enable()" is under pull request. (ARM-software/CMSIS_5#589)
I will create a pull request as soon as it is merged.

toyowata, 2019-08-01:
Fixed in 5.13.0 onwards.

closed 2019-08-01 by toyowata