ARMmbed/mbed-os #10357: Mbed os Compile for NRF52840 with JerryScript

ARM93, 2019-04-09:
###Description

I have tired to compile the mbed os 5 with jerryscript for NRF52840 target. i almost done the porting but faced issues in "tools/config/__init__.py" file "format_validation_error" function.

Lenovo-Z580:~/repel/mbed-js-repl-example$ gulp --target=NRF52840_DK
[17:56:39] Using gulpfile ~/repel/mbed-js-repl-example/Gulpfile.js
[17:56:39] Starting 'make-build-dir'...
[17:56:39] Finished 'make-build-dir' after 128 μs
[17:56:39] Starting 'makefile'...
[17:56:39] Starting 'bundle'...
[17:56:39] Starting 'ignorefile'...
[17:56:39] Finished 'makefile' after 44 ms
[17:56:39] Starting 'get-jerryscript'...
[17:56:39] Finished 'get-jerryscript' after 90 μs
[17:56:39] Starting 'getlibs'...
[17:56:39] Finished 'ignorefile' after 37 ms
[17:56:39] Finished 'getlibs' after 12 ms
$ make getlibs
make: Nothing to be done for 'getlibs'.
[17:56:39] Finished 'bundle' after 104 ms
[17:56:39] Starting 'cppify'...
[17:56:39] Finished 'cppify' after 24 ms
[17:56:39] Starting 'build'...
[17:56:40] Found native packages: mbed-js-repl
$ make BOARD=NRF52840_DK EXTRAS=/home/pc/repel/mbed-js-repl-example/node_modules/mbed-js-repl/.
[17:56:40] build/make
[17:56:40] Finished 'build' after 1.66 s
[17:56:40] Starting 'default'...
[17:56:40] Finished 'default' after 26 μs
cd jerryscript/targets/mbedos5 && pip install -r tools/requirements.txt && rm -f source/js_encoded.cpp && rm -f source/pins.cpp && rm -f source/main.cpp && make BOARD=NRF52840_DK EXTRA_SRC="/home/pc/repel/mbed-js-repl-example/node_modules/mbed-js-repl/.:../../../source" EXTERN_BUILD_DIR=../../../out/NRF52840_DK NO_JS=1
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Requirement already satisfied: pycparser<=2.17 in /usr/local/lib/python2.7/dist-packages (from -r tools/requirements.txt (line 1)) (2.17)
Requirement already satisfied: simpleeval in /usr/local/lib/python2.7/dist-packages (from -r tools/requirements.txt (line 2)) (0.9.8)
Requirement already satisfied: pycparserext in /usr/local/lib/python2.7/dist-packages (from -r tools/requirements.txt (line 3)) (2016.2)
Requirement already satisfied: ply>=3.4 in /usr/local/lib/python2.7/dist-packages (from pycparserext->-r tools/requirements.txt (line 3)) (3.11)
make[1]: Entering directory '/home/pc/repel/mbed-js-repl-example/build/jerryscript/targets/mbedos5'
python tools/generate_pins.py NRF52840_DK
mbed target NRF52840_DK
[mbed] Working path "/home/pc/repel/mbed-js-repl-example/build/jerryscript/targets/mbedos5" (library)
[mbed] Program path "/home/pc/repel/mbed-js-repl-example/build/jerryscript/targets/mbedos5"
[mbed] NRF52840_DK now set as default target in program "mbedos5"
mbed compile -j0 --source . --source ../../ --source /home/pc/repel/mbed-js-repl-example/node_modules/mbed-js-repl/. --source ../../../source --build ../../../out/NRF52840_DK -D "CONFIG_MEM_HEAP_AREA_SIZE=(1024*16)" -t GCC_ARM -v
Traceback (most recent call last):
  File "/home/pc/repel/mbed-js-repl-example/build/jerryscript/targets/mbedos5/mbed-os/tools/make.py", line 76, in wrapped_build_project
    src_dir, build_dir, mcu, *args, **kwargs
  File "/home/pc/repel/mbed-js-repl-example/build/jerryscript/targets/mbedos5/mbed-os/tools/build_api.py", line 561, in build_project
    app_config=app_config, build_profile=build_profile, ignore=ignore)
  File "/home/pc/repel/mbed-js-repl-example/build/jerryscript/targets/mbedos5/mbed-os/tools/build_api.py", line 467, in prepare_toolchain
    config = config or Config(target, src_paths, app_config=app_config)
  File "/home/pc/repel/mbed-js-repl-example/build/jerryscript/targets/mbedos5/mbed-os/tools/config/__init__.py", line 519, in __init__
    for x in errors))
  File "/home/pc/repel/mbed-js-repl-example/build/jerryscript/targets/mbedos5/mbed-os/tools/config/__init__.py", line 519, in <genexpr>
    for x in errors))
  File "/home/pc/repel/mbed-js-repl-example/build/jerryscript/targets/mbedos5/mbed-os/tools/config/__init__.py", line 461, in format_validation_error
    path, ".".join(p for p in error.absolute_path),
TypeError: sequence item 1: expected string or Unicode, int found
[mbed] ERROR: "/usr/bin/python" returned error.
       Code: 1
       Path: "/home/pc/repel/mbed-js-repl-example/build/jerryscript/targets/mbedos5"
       Command: "/usr/bin/python -u /home/pc/repel/mbed-js-repl-example/build/jerryscript/targets/mbedos5/mbed-os/tools/make.py -D CONFIG_MEM_HEAP_AREA_SIZE=(1024*16) -t GCC_ARM -m NRF52840_DK --source . --source ../../ --source /home/pc/repel/mbed-js-repl-example/node_modules/mbed-js-repl/. --source ../../../source --build ../../../out/NRF52840_DK -v -j0"
       Tip: You could retry the last command with "-v" flag for verbose output
---
[mbed] Working path "/home/pc/repel/mbed-js-repl-example/build/jerryscript/targets/mbedos5" (library)
[mbed] Program path "/home/pc/repel/mbed-js-repl-example/build/jerryscript/targets/mbedos5"
[mbed] Exec "/usr/bin/python -u /home/pc/repel/mbed-js-repl-example/build/jerryscript/targets/mbedos5/mbed-os/tools/make.py -D CONFIG_MEM_HEAP_AREA_SIZE=(1024*16) -t GCC_ARM -m NRF52840_DK --source . --source ../../ --source /home/pc/repel/mbed-js-repl-example/node_modules/mbed-js-repl/. --source ../../../source --build ../../../out/NRF52840_DK -v -j0" in "/home/pc/repel/mbed-js-repl-example/build/jerryscript/targets/mbedos5"
Makefile:56: recipe for target 'all' failed
make[1]: Leaving directory '/home/pc/repel/mbed-js-repl-example/build/jerryscript/targets/mbedos5'
make[1]: *** [all] Error 1
Makefile:14: recipe for target 'all' failed
make: *** [all] Error 2
events.js:183
      throw er; // Unhandled 'error' event
      ^

Error: Command `make BOARD=NRF52840_DK EXTRAS=/home/pc/repel/mbed-js-repl-example/node_modules/mbed-js-repl/.` exited with code 2
    at ChildProcess.handleSubShellExit (/home/pc/repel/mbed-js-repl-example/node_modules/gulp-run/command.js:166:13)
    at Object.onceWrapper (events.js:317:30)
    at emitTwo (events.js:126:13)
    at ChildProcess.emit (events.js:214:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:198:12)

Environment Details:
Target:- NRF52840
Toolchain:- arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 6.2.1 20161205 
git Source: https://github.com/ARMmbed/mbed-os.git
git SHA: 71c84e851f771fbf7d6615e415547d4574009fb0

### Issue request type

    [ ] Question
    [X] Enhancement
    [ ] Bug

ARM93, 2019-04-12:
I have done the jerryscript porting for NRF52840 chipset by modified the PinNames.h manually (i have Only configured port 0 pins in PinNames.h file. if add the port 1 pins in PinNames.h file then faced some compile time issues. So i have skipped the port 1 pins configuration in PinNames.h file) and did some patch work in definitions.json file.

After long struggle then the hex/bin file generated & it booted successfully in NRF52840 platform. By using the main.js file i can able control the LEDs. But i faced the issues in JerryScript console.

Once after the JerryScript console get opened then press any keys in keyboard (except the "Enter" key) then i faced below error.
[image]

ARM93, 2019-04-16:
I resolved this issue myself. Thanks for your support.

closed 2019-04-16 by ARM93

0xc0170, 2019-04-16:
How was this resolved @ARM93 ? Just if someone has the same issue in future, can read this and fix.