ARMmbed/mbed-os #10360: Inconsistent Licensing Unformation

maxgerhardt, 2019-04-09:
### Description

The LICENCE.md file of mbed-os states: https://github.com/ARMmbed/mbed-os/blob/master/LICENSE.md

>Folders containing files under different permissive license than Apache 2.0 are listed below. Each folder should contain its own README file with license specified for its files. The original license text is included in those source files.
> [..]
> features/storage - BSD-style, MIT

However, looking at the actual source code files of `features/storage`, they all are Apache 2.0 licensed.

https://github.com/ARMmbed/mbed-os/blob/dc1198b5c882410ea8a6b76c3b5a03dfe99eb958/features/storage/nvstore/source/nvstore.cpp#L1-L15

And no MIT/BSD licenses are to to be found there. 

Can you look over the licenses that subfolder again and correct the `LICENSE.md`?

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

maxgerhardt, 2019-04-09:
Also I don't understand https://github.com/ARMmbed/mbed-os/blob/master/features/storage/blockdevice/ExhaustibleBlockDevice.h is MIT-licensed and its source code https://github.com/ARMmbed/mbed-os/blob/master/features/storage/blockdevice/ExhaustibleBlockDevice.cpp is Apache 2.0 licensed. Huh?

maxgerhardt, 2019-04-09:
And the [LittleFS C++ class](https://github.com/ARMmbed/mbed-os/blob/master/features/storage/filesystem/littlefs/LittleFileSystem.h) is licensed Apache-2.0 whereas the [underlying C implementation](https://github.com/ARMmbed/mbed-os/blob/master/features/storage/filesystem/littlefs/littlefs/lfs.c) is in BSD-3-Clause.

cmonr, 2019-04-09:
@ARMmbed/mbed-os-storage @geky

0xc0170, 2019-04-10:
> However, looking at the actual source code files of features/storage, lots of them are also Apache 2.0 licensed.

It's stated in the license file as "Folders containing files under different permissive license than Apache 2.0 are listed below". By default, all files should be Apache 2.0.

To get detailed view, storage in this case should provide own license file with more detailed license information. This still needs to be done for many folders listed in the license file.

> Also I don't understand why header file https://github.com/ARMmbed/mbed-os/blob/master/features/storage/blockdevice/ExhaustibleBlockDevice.h is MIT-licensed and its source code https://github.com/ARMmbed/mbed-os/blob/master/features/storage/blockdevice/ExhaustibleBlockDevice.cpp is Apache 2.0 licensed. Huh?

@yossi2le Please fix

> And the LittleFS C++ class is licensed Apache-2.0 whereas the underlying C implementation is in BSD-3-Clause.

C implementation is BSD-3-Clause. The C++ object for Mbed OS is under Apache 2.0 - as stated above - this is our default license.

0xc0170, 2019-04-16:
@maxgerhardt If you got further questions, let us know.

I'll close this as resolved. Blockdevice file fix is here #10414

closed 2019-04-16 by 0xc0170