ARMmbed/mbed-os #10365: Make CRASH feature available for all targets

jeromecoutant, 2019-04-10:
### Description

Hi

Discussion started in #10349 or #10331 

Current CRASH feature is disabled for all targets:
https://github.com/ARMmbed/mbed-os/blob/ffe9ddfb2ded8e601a62d3fb1ad9adadd97f4bda/platform/mbed_lib.json#L117

and enabled for few targets in the same file.

For these targets, linker script for each tool chains have been updated to get a dedicated RAM for that CRASH DATA.

Idea is to allow each customer to enable the feature only with the json configuration management. This means that all default linker scripts have to be ready for that.
Of course, if the feature is disabled, RAM impact should be null.

@deepikabhavnani 
@SenRamakri 
@studavekar 
@mprse 
@0xc0170 

### Issue request type

    [ ] Question
    [x] Enhancement
    [ ] Bug

deepikabhavnani, 2019-04-10:
> Of course, if the feature is disabled, RAM impact should be null.

jeromecoutant, 2019-04-23:
Please reopen this issue

jeromecoutant, 2019-04-25:
@linlingao

masoudr, 2019-05-09:
Adding a new memory map for a target is not an easy task for the devs. So it would be better to just enable it in the configuration as you already stated in [here](https://github.com/ARMmbed/mbed-os/issues/10331). I'm not a newbie but it took me some hours to understand memory mapping and create the new RAM region for my new target.

On the other hand, I think occupying 256 bytes of RAM is not a big tradeoff for most targets.
For the RAM region, I think it would be better to have multiple RAM configuration for each target. Then we can link them on compile time.

0xc0170, 2019-08-13:
Agree, this would be a task for a feature branch to get all update. Ideally, it would be enabled for all but the reality is as stated, update per target basis.

cc @ARMmbed/mbed-os-core