ARMmbed/mbed-os #10370: Align linker files for all targets having multiple RAM banks

deepikabhavnani, 2019-04-10:
### Description
Targets having multiple RAM banks were not utilizing both RAM memories for all three toolchains.
Reference PR is added for all tool chains to 2 targets, which should be extended to rest of the devices.

Reference : 
ARM - https://github.com/ARMmbed/mbed-os/pull/10018
GCC - https://github.com/ARMmbed/mbed-os/pull/9944
IAR - https://github.com/ARMmbed/mbed-os/pull/10003

### Issue request type

    [ ] Question
    [X] Enhancement
    [ ] Bug

deepikabhavnani, 2019-04-10:
CC @jeromecoutant

jeromecoutant, 2019-04-23:
please reopen this issue!

jeromecoutant, 2019-04-23:
IAR update reference PR is no more #10003 but #10303

deepikabhavnani, 2019-04-23:
@ARMmbed/mbed-os-maintainers - Can we change the settings to allow the issue owner to open/close the issue?

@linlingao - Please re-open this one

linlingao, 2019-04-23:
@deepikabhavnani Can you confirm this applies to not just ST but also other targets?

deepikabhavnani, 2019-04-23:
> Can you confirm this applies to not just ST but also other targets?

Applies to all targets having multiple RAM banks

linlingao, 2019-04-23:
@deepikabhavnani Do you have a list of targets with multiple RAM banks? We should label them.

deepikabhavnani, 2019-04-23:
> Do you have a list of targets with multiple RAM banks?

I don't have the list, will have to scan all linker files / targets. Best will be to label all the partners.

jeromecoutant, 2019-04-24:
Hi

about labels:
- please remove the "devices: st"
- check jira status

Thx

linlingao, 2019-04-24:
@jeromecoutant Not sure why you requested removal of devices:st? You don't think further work is needed from your side?

jeromecoutant, 2019-04-25:
> Not sure why you requested removal of devices:st

See Deepika comment: "Applies to all targets having multiple RAM banks"
Work is ARM side.

closed 2019-10-31 by ciarmcom