ARMmbed/mbed-os #10371: Exporter support for uVision5 broken

maclobdell, 2019-04-10:
### Description

Mbed OS version: 5.12.0

Boards that previously supported exporting to uVision5 in older versions of Mbed OS, now fail with the error message

```
project.py: error: DISCO_L475VG_IOT01A not supported by uvision5
```
When issuing the following command, I see that many devices now support uVision6 but no devices support uVision5.

```
mbed export -i uvision --supported
```
[supported.txt](https://github.com/ARMmbed/mbed-os/files/3065680/supported.txt)

There seems to be a logic error in which targets that list support for "ARM" compiler are defaulting to uVision6 only, instead of both uVision5 and uVision6.

cc @theotherjimmy 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ X] Bug

0xc0170, 2019-04-11:
cc @ARMmbed/mbed-os-tools

bridadan, 2019-04-11:
I don't think this use case made it into the changes that had the ARMC5 and ARMC6, but I agree this seems like this should work in the manner that you described.

closed 2021-10-20 by ciarmcom