ARMmbed/mbed-os #10372: 'mbed dm update device' command ignores profile, can't find hex/binary in BUILD/ARMC6-RELEASE

maclobdell, 2019-04-11:
### Description
I'm testing https://github.com/ARMmbed/mbed-cloud-client-example

To get optimal codesize I set the profile to “release”.   

I used the command  ‘mbed compile --profile release’.  This generated the output in folder BUILD\LPC55S69_NS\ARMC6-RELEASE.

However, when issuing 'mbed dm update device -D <device ID>, it cannot locate the application output file.  

I discovered that I could pass in the toolchain as 'ARMC6-RELEASE' and it would find it and work correctly.  

'mbed dm update device -D <device ID> -t ARMC6-RELEASE'

My expectation is that either it magically figures out what profile I just compiled with and want to use, or there is an extra parameter needed that lets me specify it explicitly.  My guess is that option 2 is better.  

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

linlingao, 2019-08-13:
It's working in 5.13.2. The command to use is
mbed dm update device -D -m <target> --profile <profile>.

0xc0170, 2019-08-14:
Closing as resolved

Thanks @linlingao for testing

closed 2019-08-14 by 0xc0170