ARMmbed/mbed-os #10379: mbed-cli ignores profile

maxgerhardt, 2019-04-11:
I'm in the situation where I must use a custom `profiles/physec_build_profile.json` for my project due to adding a certain C++ language version and other compiler flags. 

Compiling using 

```
mbed compile --profile profiles/physec_build_release_profile.json -m NUCLEO_L476RG -t GCC_ARM
```

works fine. However, when exporting CMake files for later integration with CLion, `mbed-cli` does not respect the `--profile` option.

```
mbed export --profile profiles/physec_build_profile.json -i cmake_gcc_arm -m NUCLEO_L476RG
```

Will produce a `CMakeLists.txt` file which invokes `mbed-cli` with the wrong pfile.

```cmake
##########################################################################
# mbed-cli specific targets
##########################################################################

# detect the build type and select the corresponding cli profile
SET(MBED_BUILD_PROFILE "")
STRING(TOLOWER ${CMAKE_BUILD_TYPE} LOWERCASE_CMAKE_BUILD_TYPE)
IF(LOWERCASE_CMAKE_BUILD_TYPE MATCHES debug)
    SET(MBED_BUILD_PROFILE "mbed-os/tools/profiles/debug.json")
ELSEIF(LOWERCASE_CMAKE_BUILD_TYPE MATCHES relwithdebinfo)
    SET(MBED_BUILD_PROFILE "mbed-os/tools/profiles/develop.json")
ELSEIF(LOWERCASE_CMAKE_BUILD_TYPE MATCHES release)
    SET(MBED_BUILD_PROFILE "mbed-os/tools/profiles/release.json")
ELSEIF(LOWERCASE_CMAKE_BUILD_TYPE MATCHES minsizerel)
    SET(MBED_BUILD_PROFILE "mbed-os/tools/profiles/release.json")
ELSE()
  MESSAGE(WARNING "Build type '${CMAKE_BUILD_TYPE}' is unknown, using debug profile")
  SET(MBED_BUILD_PROFILE "mbed-os/tools/profiles/debug.json")
ENDIF()

# optional custom target to build via mbed-cli
ADD_CUSTOM_TARGET(mbed-cli-build
        COMMAND ${CMAKE_COMMAND} -E echo "mbed compile --build BUILD/${CMAKE_BUILD_TYPE} --profile ${MBED_BUILD_PROFILE}"
        COMMAND mbed compile --build BUILD/${CMAKE_BUILD_TYPE} --profile ${MBED_BUILD_PROFILE}
        WORKING_DIRECTORY ${CMAKE_CURRENT_SOURCE_DIR}
        SOURCES ${SOURCE_FILES} ${SYS_SOURCE_FILES})
``` 

You can see that it first sets `MBED_BUILD_PROFILE` to an empty string and then tries find the correct `MBED_BUILD_PROFILE` from a pre-set list. None of them are the one I specified. 

Subsequently it invokes `mbed-cli` with the `debug.json` profile and fails.

```
/home/max/Downloads/clion-2017.3/bin/cmake/linux/bin/cmake --build /home/max/Documents/my_project/cmake-build-debug --target mbed-cli-build -- -j 2
mbed compile --build BUILD/Debug --profile mbed-os/tools/profiles/debug.json
```

What is the correct way of invoking `mbed-cli` to generate a `CMakeLists.txt` which has the right `--profile`?

theotherjimmy, 2019-04-11:
@maxgerhardt Done. We're in mbed-os now.

maxgerhardt, 2019-04-18:
Any updates on this? Maybe it wasn't noticed because the issue was moved?

maxgerhardt, 2019-04-25:
@cmonr can you @ the correct team responsible for this? This is a 14 days old issue now.

maxgerhardt, 2019-05-07:
@0xc0170 Any chance to get the right team tagged in here?

cmonr, 2019-05-07:
@ARMmbed/mbed-os-tools

bridadan, 2019-05-09:
Hi @maxgerhardt, sorry for the lack of responses.

I am not very familiar with using CMake. It looks like this exporter was added by @thinkberg for use with the CLion IDE: https://github.com/ARMmbed/mbed-os/pull/5476

Perhaps you could work with @thinkberg to agree on what the behavior should be here in this case? Based on the template it looks like it was built to play nicely with the configurations provided by CLion.

thinkberg, 2019-05-10:
As far as I remember, I couldn't get profile specific details, which is why there is a compile-using-mbed-cli option in there that selects a different profile. I haven't looked at it for a while. The difficulty is that if you export for a specific profile, it will only work with that profile. CMake already has a notion of different compile settings Debug, Release etc and it would be good to get information on all different available profiles (or select them) and include them accordingly in the cmake file.