ARMmbed/mbed-os #10388: BufferedBlockDevice doesn't program the whole buffer

RobVlaar, 2019-04-12:
### Description
BufferedBlockDevice doesn't program the whole buffer. Because of the condition on line 239 only the middle part of the buffer is programmed. Removing the if statement programs all of the buffer. Writing all in once isn't possible without having an extra internal buffer.

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change

### Reviewers

davidsaada, 2019-04-15:
The whole idea of the buffered block device is to only program to the underlying block device when the end of the program unit is reached. For instance, if you have a programming unit of 8, then programming 5 bytes at the beginning of the programming unit won't perform the actual programming, but only the programming of the additional 3 bytes after that. This is the purpose of the statement at line 239. Now, if you wish to force the programming even if you didn't reach the end of the program unit, you can always call the `sync` method. However, you should only do that if you are certain that you won't add any programmed data to that program unit, or programming would fail (as you never called the `erase` method).

cmonr, 2019-04-15:
@ARMmbed/mbed-os-storage

RobVlaar, 2019-04-18:
@davidsaada thanks for the explanation, i misused the bufferedBlockDevice for my purpose.

closed 2019-04-18 by RobVlaar