ARMmbed/mbed-os #10389: Travis: keyexpired , affecting PRs today

0xc0170, 2019-04-12:
### Description

Known issue for today , few PRs are red already because of this.

```
The command "W: Failed to fetch http://dl.google.com/linux/chrome/deb/dists/stable/Release.gpg The following signatures were invalid: KEYEXPIRED 1555048520 KEYEXPIRED 1555048520 KEYEXPIRED 1555048520 KEYEXPIRED 1555048520" failed. Retrying, 2 of 3.

```

Need to wait for key updates from google 🙄 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

linlingao, 2019-04-12:
Seems fixed now. Google says "All system operational". @0xc0170 I assume you'll close it soon?

0xc0170, 2019-04-15:
Restarting few jobs now, will close if fixed. 

Thanks @linlingao

0xc0170, 2019-04-15:
fixed 💯

closed 2019-04-15 by 0xc0170