ARMmbed/mbed-os #10392: race in svcRtxMutexAcquire

culyun, 2019-04-12:
### Description

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-04-12:
Please fill in description and Issue type. Otherwise the issue will be closed

culyun, 2019-04-12:
Ok it's late a night, but I'm seeing a bug amongst the electronic sheep..  I genuinely hope I'm wrong.

Have a look at the following starting around line 194 in mbed-os/rtos/TARGET_CORTEX/rtx5/RTX/Source/rtx_mutex.c

```C++ 
 // Check if Mutex is not locked
  if (mutex->lock == 0U) {
    // Acquire Mutex
    mutex->owner_thread = thread;
    mutex->owner_next   = thread->mutex_list;
    mutex->owner_prev   = NULL;
    if (thread->mutex_list != NULL) {
        thread->mutex_list->owner_prev = mutex;
    }
    thread->mutex_list = mutex;
    mutex->lock = 1U;
    EvrRtxMutexAcquired(mutex, mutex->lock);
    status = osOK;
  } else {
  // ...
 }
```
Unless I'm mistaken there are half a dozen lines of code between checking mutex->lock and assigning mutex->lock.  If the scheduler preempts the thread trying to lock the mutex between the check in 194 and the assignment in line 204, another thread can acquire the lock.  When the original thread resumes it will:

- be unaware of the race
- corrupt the mutexe's state if still held by the other thread
- permit the caller to operate on an object that may be in an inconsistent state

BTW the semaphore implementation looks ok, since it uses rmw intrinsics.  This begs the question of why Mutex wasn't simply implemented as a specialization of a semaphore?

kjbracey-arm, 2019-04-12:
RTX issues belong over in https://github.com/ARM-software/CMSIS_5/issues, but I'll have a look at this.

Okay, I think you need to do a bit of background reading on the [theory of operation](https://www.keil.com/pack/doc/CMSIS/RTOS2/html/theory_of_operation.html).

Basically, task switches cannot happen inside the OS itself. All those svc routines are run in SVCall exception context, and as we're single-CPU, that means there's no-one else running to change any data structures, aside from interrupts.

But as mutexes are not touchable via interrupts, no explicit race protection is needed in the svcMutexXXX routines. The protection is implicit in it being in a SVCall.

On the other hand, semaphores can be released by interrupts, which is why the semaphore code would need protection when accessing the count - although the actual real semaphore work other than the count would be deferred from interrupt to postprocessing, so only the count is an issue.

A mutex is kind of a specialization of a semaphore, I guess - it adds ownership semantics and priority inheritance based on that. But another part of the specialisation is that because of the ownership, it can't be released from interrupt. Therefore it doesn't need the interrupt protection that semaphore has.

kjbracey-arm, 2019-04-12:
And to clarify how task switching works - osMutexAcquire calls svcRtxMutexAcquire by using a SVC instruction, which makes us take the SVCall exception.

That runs svcRtxMutexAcquire, and that can decide to switch task if this thread has to wait - osRtxThreadWaitEnter calls osRtxThreadSwitch, which modifies `osRtxInfo.thread.run.next`

Then svcRtxMutexAcquire returns, and heads back into the assembler SVCall veneer. That sees that osRtxInfo.thread.run.next is not equal to osRtxInfo.thread.run.curr, and does the actual context switch - it  stores the live registers into the current thread's control block, extracts the registers from the next thread's control block, and then does an exception return into that next thread.

The switches only happen due to exceptions - we take an exception from one thread and return from an exception into another. The exceptions being either SVCalls to the OS, or interrupts that trigger task switches (timer round-robin, or IRQs that signal semaphores or flags).

linlingao, 2019-04-12:
@culyun I'm closing this since Kevin has already explained. Please reopen if you still think there is a bug.

closed 2019-04-12 by linlingao