ARMmbed/mbed-os #10396: PSA code generating many ARMC6 warnings

kjbracey-arm, 2019-04-12:
### Description

PSA code is generating a lot of ARMC6 warnings. I believe these were all coming from LPC55S69_S.

[Warning] tfm_psa_api_client.c@52,27: initializing 'struct tfm_sfn_req_s *' with an expression of type 'volatile struct tfm_sfn_req_s *' discards qualifiers [-Wincompatible-pointer-types-discards-qualifiers]
[Warning] attest_iat_claims_loader.c@91,30: implicitly declaring library function 'malloc' with type 'void *(unsigned int)' [-Wimplicit-function-declaration]
[Warning] attest_iat_claims_loader.c@124,9: implicit declaration of function 'free' is invalid in C99 [-Wimplicit-function-declaration]
[Warning] attest_iat_claims_loader.c@157,12: implicit conversion from enumeration type 'enum psa_attest_err_t' to different enumeration type 'enum tfm_plat_err_t' [-Wenum-conversion]
[Warning] psa_client.h@33,17: redefinition of typedef 'psa_status_t' is a C11 feature [-Wtypedef-redefinition]
[Warning] psa_attestation_partition.c@58,30: implicitly declaring library function 'calloc' with type 'void *(unsigned int, unsigned int)' [-Wimplicit-function-declaration]
[Warning] psa_attestation_partition.c@99,13: implicit declaration of function 'tfm_panic' is invalid in C99 [-Wimplicit-function-declaration]
[Warning] error.h@32,9: 'PSA_SUCCESS' macro redefined [-Wmacro-redefined]
[Warning] its_partition.c@177,25: implicit declaration of function 'kv_init_storage_config' is invalid in C99 [-Wimplicit-function-declaration]

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-04-15:
@ARMmbed/mbed-os-psa Please review

orenc17, 2019-04-15:
Fixed most of the warnings
warnings related to PSA_SUCCESS redefinition should be fixed when #10251 would be addressed by the crypto team

0xc0170, 2019-04-30:
@orenc17 IS this ready for closing or any remaining warnings ?

alzix, 2019-05-13:
cc @Patater

closed 2019-11-18 by ciarmcom