ARMmbed/mbed-os #10399: SPI 16 bit blockwrite pads 0s

rutgervandenberg, 2019-04-14:
### Description

I'm attempting to use 16bit SPI frames. When I use the single frame write method (`virtual int write(int value);`), everything works as expected. But when I attempt to use the multi-frame read-write method (`virtual int write(const char *tx_buffer, int tx_length, char *rx_buffer, int rx_length);`) each byte in the buffer I pass is sent as a 16bit SPI frame. 
As an example in the image below, when I attempt to send the same 16bit value `0x8003` first using the single frame method, and then using the multi-frame method, the results are different.

[image]

The cause is that in `spi_master_block_write` the tx buffer is split into bytes which are passed to `spi_master_write` one by one. `spi_master_write` then sends them as 16bit frames. 

It's easily fixed, but I'm wondering if this is the intended behavior? Would a PR be welcome?

Tested on NUCLEO_F767ZI, but after a quick look this behavior seems to exist in at least multiple targets. 

Steps to reproduce: 
```
#include "mbed.h"
mbed::SPI _mbed_spi(PB_5, PB_4, PB_3);
mbed::DigitalOut cs(PA_4, 1);
int main() {
	_mbed_spi.format(16, 3);
	_mbed_spi.frequency(1e6);
	uint16_t frame = 0x8003;
	while (true) {
		Thread::wait(500);
		cs = 0;
		_mbed_spi.write(frame);
		_mbed_spi.write((char *)&frame, 2, nullptr, 0);
		cs = 1;
	}
}
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-04-15:
cc @ARMmbed/mbed-os-hal

linlingao, 2019-04-16:
@ARMmbed/team-st-mcd

rutgervandenberg, 2019-04-16:
Just to be clear - this is not an ST only issue. I happened to test it on one, but all targets I looked at (NXP LPC4088, NRF51822) during a quick review exhibit the same behavior.

linlingao, 2019-04-16:
@rutgervandenberg Thanks for the clarification. You're welcome to do a PR.

kjbracey-arm, 2019-04-17:
I believe the feature-hal-spec-spi branch is currently reworking this - @mprse