ARMmbed/mbed-os #10400: No flow-control on TARGET_LPC55S69

teetak01, 2019-04-15:
### Description

It would seem that either the board does not support flow-control, or Mbed OS is not enabling it.

https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NXP/TARGET_MCUXpresso_MCUS/TARGET_LPC55S69/TARGET_LPCXpresso/PeripheralPins.c#L65-L85

This would be required to enable stable network connectivity with AT-based ESP8266 Wi-Fi.

Related with internal ticket: https://jira.arm.com/browse/IOTHAL-483

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

teetak01, 2019-04-15:
cc: @ARMmbed/team-nxp @maclobdell

maclobdell, 2019-04-18:
@mmahadevan108  - this one is urgent for our device management team.  it will enable reliable communication to a uart-based external module.  

It looks like the **serial_set_flow_control function** is not implemented for this target.  

The definition is here
https://github.com/ARMmbed/mbed-os/blob/master/hal/serial_api.h

Also PeripheralPins.c doesn't have pins for flow control.  

```
const PinMap PinMap_UART_CTS[] = {
    {NC   , NC    , 0}
};

const PinMap PinMap_UART_RTS[] = {
    {NC   , NC    , 0}
};
```

Based on the reference manual and schematic, I believe the following pins are available on the header and support UART function with flow control using the Flexcomm 2 peripheral.

TX       P0_27
RX       P1_24
CTS     P1_26 
RTS     P1_27

mmahadevan108, 2019-04-22:
P1_26 and P1_27 are not connected to the Arduino headers

mmahadevan108, 2019-04-23:
I can provide an implementation using Flexcomm 0 (UART0) that uses the below
TX  = P1_6
RX  = P1_5
RTS = P1_7
CTS = P1_8

maclobdell, 2019-04-24:
I have a patch from Mahesh that I'm testing.  Will share results soon.

maclobdell, 2019-04-29:
Hi Mahesh, I tested your patch.  It works.  Please make a PR!

maclobdell, 2019-05-07:
The PR is merged now.  Closing.

closed 2019-05-07 by maclobdell