ARMmbed/mbed-os #10403: Watchdog example and test code

andrewc-arm, 2019-04-15:
### Description

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

Hi,

My customer is looking for Mbed-OS reference example code and test for Watchdog timer.
I searched the master tip of mbed-os but could not find any relevant API.
It seems it was introduced in 5.9 like this [document](https://os.mbed.com/docs/mbed-os/v5.9/reference/watchdog-port.html).
But I cannot find it in latest mbed-os master tip.

Could you please guide me?

Thanks,

0xc0170, 2019-04-15:
Watchdog is still on the feature branch : https://github.com/ARMmbed/mbed-os/tree/feature-watchdog

cc @donatieng

closed 2019-04-18 by ciarmcom

linlingao, 2019-04-30:
@andrewc-arm I believe your question been answered. Please reopen if you still have questions.