ARMmbed/mbed-os #10422: STM32F7 HAL_cryp API compatibility break with cube 1.15 onwards

tushartp, 2019-04-16:
### Description

### Issue request type

    [ ] Question
    [x] Enhancement
    [x] Bug

API compatibility broken with new STM32Cube_FW_F7_V1.15.0 in the mbed-os/features/mbedtls/targets/TARGET_STM/aes_alt.c file 
can we add this issue as I cannot use the latest cube software.
Below is the description from cube release notes

Rework of HAL CRYP driver (compatibility break)
- HAL CRYP driver has been redesigned with new API's, to bypass limitations on data Encryption/Decryption management present with previous HAL CRYP driver version.

- The new HAL CRYP driver is the recommended version. It is located as usual in Drivers/STM32F7xx_HAL_Driver/Src and Drivers/STM32f7xx_HAL_Driver/Inc folders. It can be enabled through switch HAL_CRYP_MODULE_ENABLED in stm32f7xx_hal_conf.h.

- The legacy HAL CRYP driver is no longer supported.

linlingao, 2019-04-17:
@Patater Can you comment on this? Which team should I assign this to, your team or ST?

Patater, 2019-04-17:
@linlingao Was Mbed OS updated to use "F7 ST CUBE V1.15.0" yet? I only see "F7 ST CUBE V1.10.0" in commit 58c4a5f83ee3 on `master`. When updating to "F7 ST CUBE V1.15.0" in Mbed OS, it looks like `mbed-os/features/mbedtls/targets/TARGET_STM/aes_alt.c` also needs to be updated. Updating the F7 ST CUBE version is generally done by ST.

linlingao, 2019-04-17:
@Patater Thanks for the comment. I'll send this to ST to assess updating F7 ST CUBE first.

linlingao, 2019-04-17:
CC @ARMmbed/team-st-mcd

jeromecoutant, 2019-10-18:
Please check #11711 
Thx

jeromecoutant, 2019-11-07:
To be closed as #11711 is merged

closed 2019-11-07 by 0xc0170