ARMmbed/mbed-os #10428: Minimum memory requirement of mbed os for new target

ganesh-ramachandran, 2019-04-17:
### Description

_**Target:**_
New Target

_**Target Memory:**_
RAM - 10K and Flash - 128K

**Issue:**
In testing of the mbed os (greentea testing), we found that many test case requirement of memory is not fulfilled by the new controller because of less RAM.
Is this expected with this RAM size ? Also, please suggest the minimum RAM and Flash requirement for mbed?

### Issue request type
    [X] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-04-17:
Mbed Enablement contains size requirements. They depend on the level, basic has different one than pelion ready. Reference https://www.mbed.com/en/about-mbed/mbed-enabled/requirements/

Baseline class should be fine for the above target (RAM - 10K and Flash - 128K). What tests do fail ?

cc @MarceloSalazar

closed 2019-04-18 by ciarmcom

ganesh-ramachandran, 2019-04-23:
> Baseline class should be fine for the above target (RAM - 10K and Flash - 128K). What tests do fail ?
> 
> cc @MarceloSalazar

@0xc0170, below linker error is issued during green tea tests compilation:
`cannot move location counter backwards (from 20002b70 to 20002400)`

linlingao, 2019-04-30:
I'd like to keep this issue closed since the linker error is tracked in 10464.